etiennedub/pyk4a #6: ValueError: Cannot set the NumPy array 'base' dependency to NULL after initialization

Jaswers, 2019-11-12:
`PS D:\Downloads\pyk4a-master\pyk4a-master\example> python .\viewer.py
[2019-11-12 14:10:08.078] [error] [t=5644] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2019-11-12 14:10:08.078] [error] [t=5644] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (359): Invalid argument to image_get_buffer(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t
[2019-11-12 14:10:08.079] [error] [t=5644] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2019-11-12 14:10:08.080] [error] [t=5644] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (397): Invalid argument to image_get_height_pixels(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t
[2019-11-12 14:10:08.080] [error] [t=5644] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2019-11-12 14:10:08.080] [error] [t=5644] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (390): Invalid argument to image_get_width_pixels(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t
Traceback (most recent call last):
  File ".\viewer.py", line 18, in <module>
    img_color = k4a.get_capture(color_only=True)
  File "C:\ProgramData\Anaconda3\lib\site-packages\pyk4a\pyk4a.py", line 65, in get_capture
    color = k4a_module.device_get_color_image()
ValueError: Cannot set the NumPy array 'base' dependency to NULL after initialization`

OS: Windows 10

Jaswers, 2019-11-12:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/555
set synchronized_images_only=True in config.py

closed 2019-11-12 by Jaswers

yamisky, 2019-11-16:
I have the same problem, but your method is useless to me.