etiennedub/pyk4a #7: ValueError: Cannot set the NumPy array 'base' dependency to NULL after initialization

yamisky, 2019-11-16:
[image]
who can help me？

lpasselin, 2019-11-16:
Could you send the script you are using or a simplified version?
Also, which bersion of pyk4a are you using?

amy-caali, 2019-11-16:
I also still get this issue if I reference the kinect image stream too many times too quickly. My solution is to restrict my frame grab rate (to about 10 frames in the space of 30 sec, I believe). I have not tried the config file fix I recently saw on another thread but would love to know how you address this. (And in the meantime, maybe my fix will limp you along too!)

lpasselin, 2019-11-16:
Are you both on the most recent version?
The recent update brings automatic releasing of the images by the python garbage collector.

I am personally using the device and the library with windows and linux to capture batches of 50 images, as fast as possible. Highest resolution at 15fps. 
We were not able to do this with the latest version because it required to copy the images.

amy-caali, 2019-11-16:
Ah, nice to hear. I’m actually using a different python wrapper entirely (brendondburns’s). I’m also using a basic laptop without a fancy graphics card (that can’t support skeleton tracking, for example), which might contribute?

lpasselin, 2019-11-16:
The two devices I use are laptops. 5 year old laptop on linux and 2 year old Windows Surface.

I don't think the original issue is related to performance.

etiennedub, 2019-12-12:
@yamisky I don't know if you still have this issue, but without more information we cannot help you on this issue.
I will close it for now, if you want you can reopen it with additional information.

closed 2019-12-12 by etiennedub

kevinkit, 2020-02-10:
I think this issue is not resolved.

I have the same error when trying the minimalistic example in the README. However, the more advanced example "viewer.py" works without problems. 

I am working on a high end machine, so it is not due to limiations of the hardware, however I am on windows which seems to be problamatic sometimes when it comes to external libraries in python etc.

Also I work with the 1.3.0 SDK not the 1.2.0 - maybe something was changed here?