etiennedub/pyk4a #12: Difficulty in installing body tracking version

dhruvapatil, 2019-12-28:
I am trying to install the body tracking feature on Windows 10 from the body tracking [branch](https://github.com/lpasselin/pyk4a/tree/feature/body-tracking-optional)  using the command: **pip install git+https://github.com/lpasselin/pyk4a.git@feature/body-tracking-optional --global-option="build_ext" --global-option="--enable-body-tracking"**
The INCLUDE and LIB paths contain the respective SDK include and lib folders. However, I am getting the following error:
[image]

Can someone please help me with this issue?
Thank you!

lpasselin, 2019-12-28:
You can try again, I removed the preprocessor message.

dhruvapatil, 2019-12-28:
Thank you for the prompt reply @lpasselin!
I tried installing it again, and got the following error:

**pyk4a/pyk4a.cpp(18): fatal error C1016: #if[n]def expected an identifier
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.24.28314\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2**

I had previously tried adding __cplusplus extern "C" { to my local branch at that line in pyk4a.cpp but it wasn't the right TAG expected for the if[n]def .

lpasselin, 2019-12-30:
Thanks for the errors, I fixed my branch again.
Using preprocessor like I did is not clean.

dhruvapatil, 2019-12-30:
Thanks for the timely updates! I pulled the new branch again, and the preprocessor error is fixed. It seems there is a syntax error 2059 with the assignment for the k4abt_get_num_bodies on line 234. I have been trying hard to solve it, but my knowledge of C++ is limited. Apologies for constantly bugging you with the error reports.
 **pyk4a/pyk4a.cpp(234): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(234): error C2059: syntax error: ']'
    pyk4a/pyk4a.cpp(235): error C2371: 'num_bodies': redefinition; different basic types
    pyk4a/pyk4a.cpp(233): note: see declaration of 'num_bodies'
    pyk4a/pyk4a.cpp(236): error C2374: 'buffer': redefinition; multiple initialization
    pyk4a/pyk4a.cpp(234): note: see declaration of 'buffer'
    pyk4a/pyk4a.cpp(236): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(236): error C2059: syntax error: ']'
    pyk4a/pyk4a.cpp(241): warning C4267: 'argument': conversion from 'size_t' to 'uint32_t', possible loss of data
    pyk4a/pyk4a.cpp(254): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(254): error C2143: syntax error: missing ')' before ';'
    pyk4a/pyk4a.cpp(254): error C2059: syntax error: ')'
    pyk4a/pyk4a.cpp(254): error C2143: syntax error: missing ';' before ']'
    pyk4a/pyk4a.cpp(254): error C2059: syntax error: ']'
    pyk4a/pyk4a.cpp(255): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(255): error C2143: syntax error: missing ')' before ';'
    pyk4a/pyk4a.cpp(255): error C2059: syntax error: ')'
    pyk4a/pyk4a.cpp(255): error C2143: syntax error: missing ';' before ']'
    pyk4a/pyk4a.cpp(255): error C2059: syntax error: ']'
    pyk4a/pyk4a.cpp(257): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(257): error C2143: syntax error: missing ')' before ';'
    pyk4a/pyk4a.cpp(257): error C2059: syntax error: ')'
    pyk4a/pyk4a.cpp(257): error C2143: syntax error: missing ';' before ']'
    pyk4a/pyk4a.cpp(257): error C2059: syntax error: ']'
    pyk4a/pyk4a.cpp(258): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(258): error C2143: syntax error: missing ')' before ';'
    pyk4a/pyk4a.cpp(258): error C2059: syntax error: ')'
    pyk4a/pyk4a.cpp(258): error C2143: syntax error: missing ';' before ']'
    pyk4a/pyk4a.cpp(258): error C2059: syntax error: ']'
    pyk4a/pyk4a.cpp(259): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(259): error C2143: syntax error: missing ')' before ';'
    pyk4a/pyk4a.cpp(259): error C2059: syntax error: ')'
    pyk4a/pyk4a.cpp(259): error C2143: syntax error: missing ';' before ']'
    pyk4a/pyk4a.cpp(259): error C2059: syntax error: ']'
    pyk4a/pyk4a.cpp(262): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(262): error C2143: syntax error: missing ')' before ';'
    pyk4a/pyk4a.cpp(262): error C2059: syntax error: ')'
    pyk4a/pyk4a.cpp(262): error C2143: syntax error: missing ';' before ']'
    pyk4a/pyk4a.cpp(262): error C2059: syntax error: ']'
    pyk4a/pyk4a.cpp(263): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(263): error C2143: syntax error: missing ')' before ';'
    pyk4a/pyk4a.cpp(263): error C2059: syntax error: ')'
    pyk4a/pyk4a.cpp(263): error C2143: syntax error: missing ';' before ']'
    pyk4a/pyk4a.cpp(263): error C2059: syntax error: ']'
    pyk4a/pyk4a.cpp(264): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(264): error C2143: syntax error: missing ')' before ';'
    pyk4a/pyk4a.cpp(264): error C2059: syntax error: ')'
    pyk4a/pyk4a.cpp(264): error C2143: syntax error: missing ';' before ']'
    pyk4a/pyk4a.cpp(264): error C2059: syntax error: ']'
    pyk4a/pyk4a.cpp(265): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(265): error C2143: syntax error: missing ')' before ';'
    pyk4a/pyk4a.cpp(265): error C2059: syntax error: ')'
    pyk4a/pyk4a.cpp(265): error C2143: syntax error: missing ';' before ']'
    pyk4a/pyk4a.cpp(265): error C2059: syntax error: ']'
    pyk4a/pyk4a.cpp(268): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(268): error C2143: syntax error: missing ')' before ';'
    pyk4a/pyk4a.cpp(268): error C2059: syntax error: ')'
    pyk4a/pyk4a.cpp(268): error C2143: syntax error: missing ';' before ']'
    pyk4a/pyk4a.cpp(268): error C2059: syntax error: ']'
    pyk4a/pyk4a.cpp(276): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(277): error C2059: syntax error: '='
    pyk4a/pyk4a.cpp(280): error C2065: 'pose_capsule_cleanup': undeclared identifier
    pyk4a/pyk4a.cpp(328): error C2059: syntax error: '{'
    pyk4a/pyk4a.cpp(328): error C2143: syntax error: missing ';' before '{'
    pyk4a/pyk4a.cpp(328): error C2447: '{': missing function header (old-style formal list?)
    pyk4a/pyk4a.cpp(328): error C2059: syntax error: ','
    pyk4a/pyk4a.cpp(330): error C2143: syntax error: missing ';' before '{'
    pyk4a/pyk4a.cpp(330): error C2447: '{': missing function header (old-style formal list?)
    pyk4a/pyk4a.cpp(381): error C2059: syntax error: '}'
    pyk4a/pyk4a.cpp(381): error C2143: syntax error: missing ';' before '}'**

lpasselin, 2019-12-30:
Ok I finally took time to compile it on my end. With the last fixes it compiles fine. I confirmed the pip command also works if you have libk4abt in the path.

dhruvapatil, 2019-12-30:
I tried installing again, and got the same list of errors. I suspect I might be having a different Visual Studio version on my Windows 10 system. What version of Visual Studio are you using on Windows 10?
Thanks again for your timely replies!

lpasselin, 2019-12-30:
I am not on windows, I use the most recent gpp compiler on linux.

I just realized my git push did not finish. Sorry about that. Can you try again?

dhruvapatil, 2019-12-30:
The installation was a success and the test script for 2d skeleton works perfectly! Thanks a lot for patiently helping me with the installation. The body tracking feature will be a useful addition to the python wrapper for Kinect.

closed 2019-12-30 by dhruvapatil