etiennedub/pyk4a #14: k4a_module did not found

abbott1234, 2020-02-18:
GOT AN ERROR :Traceback (most recent call last):
  File "viewer.py", line 1, in <module>
    import pyk4a
  File "D:\ProgramData\Anaconda3\envs\pose\lib\site-packages\pyk4a\__init__.py", line 2, in <module>
    from .pyk4a import *
  File "D:\ProgramData\Anaconda3\envs\pose\lib\site-packages\pyk4a\pyk4a.py", line 2, in <module>
    import k4a_module
ImportError: DLL load failed:  cannot find the specified module 

fristly, I think it is the SDK vision problem, so I was trying to find vision1.2 SDK. but when I was installed vision1.2 it did not worked either.

lpasselin, 2020-02-18:
Could you make sure the required modules are available in the PATH variable?

See [specific instructions for windows in the readme](https://github.com/etiennedub/pyk4a#windows). Adapt with your own paths and versions.

abbott1234, 2020-02-19:
I have set those paths to my env :LINK="C:\Program Files\Azure Kinect SDK v1.3.0\sdk\windows-desktop\amd64\release\lib\k4a.lib"
Link-short="C:\Program Files\Azure Kinect SDK v1.3.0\sdk\windows-desktop\amd64\release\bin"
link1="C:\Program Files\Azure Kinect SDK v1.3.0\sdk\windows-desktop\amd64\release\lib"
link2="C:\Program Files\Azure Kinect SDK v1.3.0\sdk\windows-desktop\amd64\release\"
link3="C:\Program Files\Azure Kinect SDK v1.3.0\sdk\windows-desktop\amd64\release\lib\k4a.lib"
link4="C:\Program Files\Azure Kinect SDK v1.2.0\sdk\windows-desktop\amd64\release\lib\k4a.lib"
link5="C:\Program Files\Azure Kinect SDK v1.2.0\sdk\windows-desktop\amd64\release\lib\"
link6="C:\Program Files\Azure Kinect SDK v1.2.0\sdk\windows-desktop\amd64\release\bin"

And I have tried with quotes or without quotes, did not work either.

abbott1234, 2020-02-19:
The last error is solved by accident, I just try import numpy first and then import pyk4a, it is worked then without change any path env.

But viewer.py get another error, shows below:
[2020-02-19 13:56:33.959] [error] [t=2644] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\dynlib\dynlib_windows.c (120): dynlib_create(). Failed to load DLL depthengine_2_0 with error code: 126
[2020-02-19 13:56:33.959] [error] [t=2644] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\deloader\deloader.c (70): load_depth_engine(). Failed to Load Depth Engine Plugin (depthengine). Depth functionality will not work
[2020-02-19 13:56:33.960] [error] [t=2644] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\deloader\deloader.c (71): load_depth_engine(). Make sure the depth engine plugin is in your loaders path
[2020-02-19 13:56:33.960] [error] [t=2644] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\deloader\deloader.c (178): deloader_transform_engine_create_and_initialize(). Failed to load depth engine plugin
[2020-02-19 13:56:33.960] [error] [t=2644] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\tewrapper\tewrapper.c (61): transform_engine_start_helper(). Transform engine create and initialize failed with error code: 108.
[2020-02-19 13:56:33.961] [error] [t=2644] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\tewrapper\tewrapper.c (68): teresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in transform_engine_start_helper()
[2020-02-19 13:56:33.961] [error] [t=2644] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\tewrapper\tewrapper.c (86): transform_engine_start_helper(tewrapper) returned failure in transform_engine_thread()
[2020-02-19 13:56:33.961] [error] [t=16168] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\tewrapper\tewrapper.c (313): tewrapper_create(). Transform Engine thread failed to start
[2020-02-19 13:56:33.964] [error] [t=16168] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\transformation\transformation.c (433): transformation_context->tewrapper != NULL returned failure in transformation_create()
Traceback (most recent call last):
  File "viewer.py", line 10, in <module>
    k4a.connect()
  File "D:\ProgramData\Anaconda3\envs\azure\lib\site-packages\pyk4a\pyk4a.py", line 37, in connect
    self._start_cameras()
  File "D:\ProgramData\Anaconda3\envs\azure\lib\site-packages\pyk4a\pyk4a.py", line 55, in _start_cameras
    self._verify_error(res)
  File "D:\ProgramData\Anaconda3\envs\azure\lib\site-packages\pyk4a\pyk4a.py", line 184, in _verify_error
    raise K4AException()
pyk4a.pyk4a.K4AException

Azure Kinect Viewer v1.2.0 or Azure Kinect Viewer v1.3.0 both works well.

lpasselin, 2020-02-19:
I am sorry, I am not used to windows paths.

It looks like it cannot find the kinect depthengine dll. Can you make sure it can be found?

abbott1234, 2020-02-20:
I have thought about this, but I did not understand how did python link to depthengine dll, either did k4a.dll. Can you tell me which file and which line link them?

lpasselin, 2020-02-20:
The Kinect azure needs the depthengine ddl at runtime. It can be installed
with the regular installer.

Sorry again, I am no good with windows paths.

closed 2020-03-11 by abbott1234

lpasselin, 2020-09-12:
@WestbrookZero did you find a solution to this problem?