etiennedub/pyk4a #17: calibration_3d_to_3d and more

omartin2010, 2020-03-05:
guys, would you be interested in getting a PR in for supporting the coordinates transformation functions? I've just completed the work + additional class for Calibration (basic version of the C++ version of the Calibration class - I need to rework the hierarchy once I figure things out better). I've completed what I think is a working versino of calibration_3d_to_3d for now. Probably not optimized at this point but it looks like it's working.

omartin2010, 2020-03-06:
got the 2d_to_3d too, now that I realized that's the one I really needed for my project :). I'll pull those in soon, happy to go through code review with real experts :)

lpasselin, 2020-03-06:
https://github.com/etiennedub/pyk4a/pull/18

closed 2020-03-06 by lpasselin