etiennedub/pyk4a #21: Transform to point cloud data

abbott1234, 2020-03-11:
How to using this package to get point cloud?
I can only get depth image and color image

lpasselin, 2020-03-11:
This feature is not implemented yet. It should not be hard to implement. The transformation to use is  [k4a_transformation_depth_image_to_point_cloud](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga7385eb4beb9d8892e8a88cf4feb3be70.html#ga7385eb4beb9d8892e8a88cf4feb3be70) and you can follow the logic of the pyk4a.cpp function [transformation_depth_image_to_color_camera](https://github.com/etiennedub/pyk4a/blob/master/pyk4a/pyk4a.cpp#L226)

Please propose a PR with this feature if you implement it!

abbott1234, 2020-03-12:
I did not understand how does pyk4a link to Azure Kinect Sensor SDK.
when i was reading pyk4a.cpp and find a function named 'device_get_calibration' in line 150, but I can not call this function with code 'k4a_module.device_get_calibration'.

lpasselin, 2020-06-11:
I think you were using the wrong code (pypi is version 0.3, which is not the master branch of this repository)

Feel free to reopen or start a pull request with the point cloud data feature.

closed 2020-06-11 by lpasselin