etiennedub/pyk4a #24: Cannot install pyk4a

robinfays12, 2020-04-12:
Hello,

When I'm using this command from your code in Jupyterlab :

pip install pyk4a --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.3.0\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.3.0\sdk\windows-desktop\amd64\release\lib"

I have this answer :
[image]

Do you know what can be the problem ? I added the folder to my path variables like this :
[image]

Thanks for your answer !

Robin Fays

lpasselin, 2020-04-13:
Hi, the error message says to use -std=c++11 did you try this?

robinfays12, 2020-04-13:
Hi, do you know how to do it ?

lpasselin, 2020-04-13:
Can you try to add `--global-option="-std=c++11"` to the install command?

Otherwise on linux iirc I set the envvariable `export CPPFLAGS="-std=gnu++11"` before installing.

robinfays12, 2020-04-13:
I did it, and I had this answer (I'm working on Windows):

[image]

lpasselin, 2020-04-13:
I updated my comment above, can you try that?

robinfays12, 2020-04-14:
Do you mean to try this one ?

[image]

Is it on my terminal ?

lpasselin, 2020-04-14:
The export command will not work on windows. I don't know the alternative but I am sure there is one.

As for the other solution in the linked stack overflow answer, you would need to modify the setup.py extra_compile_args

robinfays12, 2020-04-14:
Ok, thank you !

closed 2020-04-16 by lpasselin