etiennedub/pyk4a #25: Unable to install on Linux

neixlo, 2020-04-14:
At first: Thanks for making this repo available!

I'm currently unable to install pyk4a on linux (Mint 19.1 which is based on Ubuntu 18.04).

**Problem:**
It seems that the compiler can't find the k4a.h.

```
pyk4a/pyk4a.cpp:4:10: fatal error: k4a/k4a.h: No such file or directory
     #include <k4a/k4a.h>
              ^~~~~~~~~~~
    compilation terminated.
    error: command 'gcc' failed with exit status 1
```

**What i did:**

1. Installed the Azure-Kinect-Sensor-SDK 1.4.0 according to the instruction.
2. Tested the ./bin/k4aviewer --> worked 
3. Added the LD_LIBRARY_PATH (At least I think I did..)

```
(Kinect_env) nixi@nixis-laptop:~/Projects/Azure_Kinect/pyk4a/example$ export LD_LIBRARY_PATH=/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/
(Kinect_env) nixi@nixis-laptop:~/Projects/Azure_Kinect/pyk4a/example$ echo $LD_LIBRARY_PATH
/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/
```

4. run: pip install pyk4a

```
(Kinect_env) nixi@nixis-laptop:~/Projects/Azure_Kinect/pyk4a/example$ pip install pyk4a
Collecting pyk4a
  Using cached pyk4a-0.3.tar.gz (6.9 kB)
Requirement already satisfied: numpy in /home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages (from pyk4a) (1.18.2)
Building wheels for collected packages: pyk4a
  Building wheel for pyk4a (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/nixi/anaconda3/envs/Kinect_env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-12goisa7/pyk4a/setup.py'"'"'; __file__='"'"'/tmp/pip-install-12goisa7/pyk4a/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-457cvchg
       cwd: /tmp/pip-install-12goisa7/pyk4a/
  Complete output (27 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.8
  creating build/lib.linux-x86_64-3.8/pyk4a
  copying pyk4a/pyk4a.py -> build/lib.linux-x86_64-3.8/pyk4a
  copying pyk4a/config.py -> build/lib.linux-x86_64-3.8/pyk4a
  copying pyk4a/__init__.py -> build/lib.linux-x86_64-3.8/pyk4a
  running build_ext
  building 'k4a_module' extension
  creating build/temp.linux-x86_64-3.8
  creating build/temp.linux-x86_64-3.8/pyk4a
  gcc -pthread -B /home/nixi/anaconda3/envs/Kinect_env/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages/numpy/core/include -I/home/nixi/anaconda3/envs/Kinect_env/include/python3.8 -c pyk4a/pyk4a.cpp -o build/temp.linux-x86_64-3.8/pyk4a/pyk4a.o
  cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
  In file included from /home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages/numpy/core/include/numpy/ndarraytypes.h:1832:0,
                   from /home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages/numpy/core/include/numpy/ndarrayobject.h:12,
                   from /home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages/numpy/core/include/numpy/arrayobject.h:4,
                   from pyk4a/pyk4a.cpp:2:
  /home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
   #warning "Using deprecated NumPy API, disable it with " \
    ^~~~~~~
  pyk4a/pyk4a.cpp:4:10: fatal error: k4a/k4a.h: No such file or directory
   #include <k4a/k4a.h>
            ^~~~~~~~~~~
  compilation terminated.
  error: command 'gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for pyk4a
  Running setup.py clean for pyk4a
[... 154 lines cut ...]
    In file included from /usr/include/c++/7/cstddef:50:0,
                     from /home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/k4a/k4atypes.h:12,
                     from /home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/k4a/k4a.h:16,
                     from pyk4a/pyk4a.cpp:4:
    pyk4a/pyk4a.cpp:231:26: error: return-statement with a value, in function returning 'void' [-fpermissive]
     #define INITERROR return NULL
                              ^
    pyk4a/pyk4a.cpp:249:13: note: in expansion of macro ‘INITERROR’
                 INITERROR;
                 ^~~~~~~~~
    pyk4a/pyk4a.cpp:251:16: error: return-statement with a value, in function returning 'void' [-fpermissive]
             return module;
                    ^~~~~~
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-hqYnRx/pyk4a/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" build_ext -I/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/ -L/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/ install --record /tmp/pip-8stf3r-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-hqYnRx/pyk4a/
(Kinect_env) nixi@nixis-laptop:~/Projects/Azure_Kinect/pyk4a/example$ 
```
I did also run it as root, same outcome.

The same 5 steps I did on my Jetson Nano which has no Conda environment running, a different architecture and a linux system derived from Ubuntu.
There the error massages are alike to the try above.

I would really appreciate help on that.

omartin2010, 2020-04-14:
if that helps, I have a dockerfile [here ](https://github.com/omartin2010/jetson_object_detection/tree/master/docker) that works with this... you may have to change it a bit because I was using prerelease code. It's also not on a Nano but on a Jetson TX2. But essentially you'd have to change a few lines in the Dockerfile, AFAIK, to get that working. My image includes opencv & tensorflow, both of which work well with the k4a. Hope this helps... at least running it in Docker and maybe help troubleshoot your issue.

lpasselin, 2020-04-14:
I think the LD_LIBRARY_PATH addition is not the problem here. You are not able to find the header file. In your case it requires additional includes. Possibly because the k4a SDK header files were not added to the usual  folders that gcc uses in the search path for headers (for example /usr/include).

Step 5 won't work because your system python is 2.7

You should try again step 4 without sudo but with the global-options includes. The compiler must be able to see the k4a header files.

Also, if you seem to be using an anaconda environment. I am not sure if this environment's gcc will be using typical system include folders or anaconda custom folders. Watch out for this.

neixlo, 2020-04-15:
@omartin2010 and @lpasselin: Thanks for the fast and supportive replay!

Unfortunately it's still not working.

I tried as suggested, without sudo and with global build options: 

```
(Kinect_env) nixi@nixis-laptop:~/Projects/Azure_Kinect/pyk4a$ pip install --global-option build_ext --global-option="-I/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/" --global-option="-L/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/" -e /home/nixi/Projects/Azure_Kinect/pyk4a
/home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages/pip/_internal/commands/install.py:244: UserWarning: Disabling all use of wheels due to the use of --build-option / --global-option / --install-option.
  cmdoptions.check_install_build_global(options)
Obtaining file:///home/nixi/Projects/Azure_Kinect/pyk4a
Requirement already satisfied: numpy in /home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages (from pyk4a==0.3) (1.18.2)
Installing collected packages: pyk4a
  Running setup.py develop for pyk4a
    ERROR: Command errored out with exit status 1:
     command: /home/nixi/anaconda3/envs/Kinect_env/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/nixi/Projects/Azure_Kinect/pyk4a/setup.py'"'"'; __file__='"'"'/home/nixi/Projects/Azure_Kinect/pyk4a/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' build_ext -I/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/ -L/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/ develop --no-deps
         cwd: /home/nixi/Projects/Azure_Kinect/pyk4a/
    Complete output (19 lines):
    running build_ext
    building 'k4a_module' extension
    creating build
    creating build/temp.linux-x86_64-3.8
    creating build/temp.linux-x86_64-3.8/pyk4a
    gcc -pthread -B /home/nixi/anaconda3/envs/Kinect_env/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages/numpy/core/include -I/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/ -I/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/ -I/home/nixi/anaconda3/envs/Kinect_env/include/python3.8 -c pyk4a/pyk4a.cpp -o build/temp.linux-x86_64-3.8/pyk4a/pyk4a.o
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
    In file included from /home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages/numpy/core/include/numpy/ndarraytypes.h:1832:0,
                     from /home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages/numpy/core/include/numpy/ndarrayobject.h:12,
                     from /home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages/numpy/core/include/numpy/arrayobject.h:4,
                     from pyk4a/pyk4a.cpp:2:
    /home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
     #warning "Using deprecated NumPy API, disable it with " \
      ^~~~~~~
    creating build/lib.linux-x86_64-3.8
    g++ -pthread -shared -B /home/nixi/anaconda3/envs/Kinect_env/compiler_compat -L/home/nixi/anaconda3/envs/Kinect_env/lib -Wl,-rpath=/home/nixi/anaconda3/envs/Kinect_env/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.8/pyk4a/pyk4a.o -L/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/ -L/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/ -lk4a -o build/lib.linux-x86_64-3.8/k4a_module.cpython-38-x86_64-linux-gnu.so
    /home/nixi/anaconda3/envs/Kinect_env/compiler_compat/ld: cannot find -lk4a
    collect2: error: ld returned 1 exit status
    error: command 'g++' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/nixi/anaconda3/envs/Kinect_env/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/nixi/Projects/Azure_Kinect/pyk4a/setup.py'"'"'; __file__='"'"'/home/nixi/Projects/Azure_Kinect/pyk4a/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' build_ext -I/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/ -L/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/ develop --no-deps Check the logs for full command output.
(Kinect_env) nixi@nixis-laptop:~/Projects/Azure_Kinect/pyk4a$
```

And also without active Anaconda environment:

```
nixi@nixis-laptop:~/Projects/Azure_Kinect/pyk4a$ pip install --global-option build_ext --global-option="-I/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/bui [... cut ...]
  File "<stdin>", line 1, in <module>
  File "/home/nixi/Projects/Azure_Kinect/pyk4a/pyk4a/__init__.py", line 2, in <module>
    from .pyk4a import *
  File "/home/nixi/Projects/Azure_Kinect/pyk4a/pyk4a/pyk4a.py", line 2, in <module>
    import k4a_module
ImportError: /home/nixi/Projects/Azure_Kinect/pyk4a/k4a_module.cpython-38-x86_64-linux-gnu.so: undefined symbol: k4a_device_open
>>> 
```
Any recomandation on how to make my compiler see and work correctly with the header file?
Or any workaround would also fit me..

Thanks again!

lpasselin, 2020-04-16:
I just noticed you are not properly adding the k4a lib folder with global-option. The global option starting with "-L" should be importing folders ending in "lib", not "include". This might be why it cannot find the k4a library.

Can you try that last import command without being in the pyk4a folder? I have a feeling the install works but you were importing the local folder.

lpasselin, 2020-04-16:
I just realized your LD_LIBRARY_PATH (from first post) could be wrong. Make sure the k4a SDK path you are adding is a folder containing the file `libk4a.so` (this folder usually ends with "lib", not "include"). It will be in your locally build folder after compilation of the sdk.

If you used the .deb by microsoft to install libk4a, maybe the lib name is different (libk4a1.4.so ? not sure). You would need to create a symlink to libk4a.so in a folder included in your LD_LIBRARY_PATH

If you fix this, install should be able to find the k4a library and you can try again installing in your anaconda environment with global option to specify the header files (the global-option starting with `-I`). (No need for the -e local install).

neixlo, 2020-04-16:
Okay, I just checked that.

So in my Azure Kinect installation is no lib folder with an `lib4a.so` or  any related named file.
I found them in `.../Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/bin` and they are already linked (`lib4a.so` is linked to `lib4a.so.1.4` and that is linked to `lib4a.so.1.4.0`.

[image]

1. I added the path of the bin folder to  LD_LIBRARY_PATH in anaconda env and also outside 
2. Then I additionaly added the path to the global option starting with "-L" and didn't refer to the local version with -e.
3. Tried to install without conda env. --> Failed
4.  Tried to install in the conda env. --> Install succeded but fails again when importing pyk4a in python.

Output to No.3:
```
nixi@nixis-laptop:~/Projects/Azure_Kinect/pyk4a$ export LD_LIBRARY_PATH=/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/bin
nixi@nixis-laptop:~/Projects/Azure_Kinect/pyk4a$ $LD_LIBRARY_PATH
bash: /home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/bin: No such file or directory
nixi@nixis-laptop:~/Projects/Azure_Kinect/pyk4a$ pip install --global-option build_ext --global-option="-I/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/" --global-option="-L/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/bin" pyk4a
/usr/lib/python2.7/dist-packages/pip/commands/install.py:212: UserWarning: Disabling all use of wheels due to the use of --build-options / --global-options / --install-options.
  cmdoptions.check_install_build_global(options)
Collecting pyk4a
  Using cached https://files.pythonhosted.org/packages/ab/95/39fc624670d1f2c303e8db27b92a8d391085bcdb654004fcba0a43f41e7a/pyk4a-0.3.tar.gz
Collecting numpy (from pyk4a)
  Using cached https://files.pythonhosted.org/packages/b7/6f/24647f014eef9b67a24adfcbcd4f4928349b4a0f8393b3d7fe648d4d2de3/numpy-1.16.6.zip
Skipping bdist_wheel for pyk4a, due to binaries being disabled for it.
Skipping bdist_wheel for numpy, due to binaries being disabled for it.
Installing collected packages: numpy, pyk4a
  Running setup.py install for numpy ... done
  Running setup.py install for pyk4a ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-1YKK60/pyk4a/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" build_ext -I/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/ -L/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/:/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/bin install --record /tmp/pip-i4AOKX-record/install-record.txt --single-version-externally-managed --compile --user --prefix=:
    running build_ext
    building 'k4a_module' extension
    creating build
    creating build/temp.linux-x86_64-2.7
    creating build/temp.linux-x86_64-2.7/pyk4a
    x86_64-linux-gnu-gcc -pthread -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-5Z483E/python2.7-2.7.17=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/home/nixi/.local/lib/python2.7/site-packages/numpy/core/include -I/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/include/ -I/home/nixi/Projects/Azure_Kinect/Azure-Kinect-Sensor-SDK/build/src/sdk/include/ -I/usr/include/python2.7 -c pyk4a/pyk4a.cpp -o build/temp.linux-x86_64-2.7/pyk4a/pyk4a.o
    In file included from /home/nixi/.local/lib/python2.7/site-packages/numpy/core/include/numpy/ndarraytypes.h:1822:0,
                     from /home/nixi/.local/lib/python2.7/site-packages/numpy/core/include/numpy/ndarrayobject.h:12,
                     from /home/nixi/.local/lib/python2.7/site-packages/numpy/core/include/numpy/arrayobject.h:4,
                     from pyk4a/pyk4a.cpp:2:
    /home/nixi/.local/lib/python2.7/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
     #warning "Using deprecated NumPy API, disable it with " \
      ^~~~~~~
    In file included from /usr/include/python2.7/Python.h:84:0,
                     from pyk4a/pyk4a.cpp:1:
    pyk4a/pyk4a.cpp: In function ‘int pyk4a_traverse(PyObject*, visitproc, void*)’:
    pyk4a/pyk4a.cpp:187:44: error: ‘PyModule_GetState’ was not declared in this scope
     #define GETSTATE(m) ((struct module_state*)PyModule_GetState(m))
                                                ^
    /usr/include/python2.7/objimpl.h:341:13: note: in definition of macro ‘Py_VISIT’
             if (op) {                                                       \
                 ^~
    pyk4a/pyk4a.cpp:209:18: note: in expansion of macro ‘GETSTATE’
             Py_VISIT(GETSTATE(m)->error);
[... cut ...]
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyk4a
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/nixi/Projects/Azure_Kinect/pyk4a/pyk4a/__init__.py", line 2, in <module>
    from .pyk4a import *
  File "/home/nixi/Projects/Azure_Kinect/pyk4a/pyk4a/pyk4a.py", line 2, in <module>
    import k4a_module
ImportError: /home/nixi/Projects/Azure_Kinect/pyk4a/k4a_module.cpython-38-x86_64-linux-gnu.so: undefined symbol: k4a_device_open
>>> 

```

Did I do everything as you suggested?

neixlo, 2020-04-16:
The same as No.3 is also failing on the Jetson Nano as well.

...
```
                     from /home/nixi/Downloads/Jetson_Setup/Azure-Kinect-Sensor-SDK/include/k4a/k4a.h:16,
                     from pyk4a/pyk4a.cpp:4:
    pyk4a/pyk4a.cpp:231:26: error: return-statement with a value, in function returning 'void' [-fpermissive]
     #define INITERROR return NULL
                              ^
    pyk4a/pyk4a.cpp:249:13: note: in expansion of macro ‘INITERROR’
                 INITERROR;
                 ^~~~~~~~~
    pyk4a/pyk4a.cpp:251:16: error: return-statement with a value, in function returning 'void' [-fpermissive]
             return module;
                    ^~~~~~
    error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-baW2kn/pyk4a/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" build_ext -I/home/nixi/Downloads/Jetson_Setup/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Downloads/Jetson_Setup/Azure-Kinect-Sensor-SDK/build/src/sdk/include/ -L/home/nixi/Downloads/Jetson_Setup/Azure-Kinect-Sensor-SDK/include/:/home/nixi/Downloads/Jetson_Setup/Azure-Kinect-Sensor-SDK/build/src/sdk/include/:/home/nixi/Downloads/Jetson_Setup/Azure-Kinect-Sensor-SDK/build/bin install --record /tmp/pip-41EV0J-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-baW2kn/pyk4a/
nixi@nixis-nano:~$ 

```

lpasselin, 2020-04-16:
1) whenever you use python 2.7, it will not build. You need to understand that your system python is version 2.7. run `python --version` to confirm you are not using python version 2.X

2) Try again in anaconda environment but in an other folder. the `import pyk4a` is importing your current working directory pyk4a folder. You do not want this. You can to import the pip installed folder which has the k4a_module linked. 

I will close this issue as this is not a pyk4a problem. It's an environment configuration problem on your system. I will still continue to help though.

closed 2020-04-16 by lpasselin

neixlo, 2020-04-16:
1.  Got it, my bad.

2. I understand and i tried:
```
(Kinect_env) nixi@nixis-laptop:~$ python
Python 3.8.2 (default, Mar 26 2020, 15:53:00) 
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyk4a
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages/pyk4a/__init__.py", line 2, in <module>
    from .pyk4a import *
  File "/home/nixi/anaconda3/envs/Kinect_env/lib/python3.8/site-packages/pyk4a/pyk4a.py", line 2, in <module>
    import k4a_module
ImportError: libk4a.so.1.4: cannot open shared object file: No such file or directory
>>> 
```
Thanks for the support.
I'll need to figure out why its not able to open `libk4a.so.1.4`

neixlo, 2020-04-16:
**It's working now!**

I needed to add the .../bin path which contains `libk4a.so.1.4` again to LD_LIBRARY_PATH.
After installation it appeared to be empty.