etiennedub/pyk4a #28: Unable to install pyk4a using pip3

vnt1537, 2020-05-05:
The error I receive is :

Collecting pyk4a
  Downloading https://files.pythonhosted.org/packages/ab/95/39fc624670d1f2c303e8db27b92a8d391085bcdb654004fcba0a43f41e7a/pyk4a-0.3.tar.gz
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from pyk4a)
Building wheels for collected packages: pyk4a
  Running setup.py bdist_wheel for pyk4a ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-2pf4lz14/pyk4a/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp0xd5qmq2pip-wheel- --python-tag cp36:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-aarch64-3.6
  creating build/lib.linux-aarch64-3.6/pyk4a
  copying pyk4a/pyk4a.py -> build/lib.linux-aarch64-3.6/pyk4a
  copying pyk4a/config.py -> build/lib.linux-aarch64-3.6/pyk4a
  copying pyk4a/__init__.py -> build/lib.linux-aarch64-3.6/pyk4a
  running build_ext
  building 'k4a_module' extension
  creating build/temp.linux-aarch64-3.6
  creating build/temp.linux-aarch64-3.6/pyk4a
  aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/lib/python3/dist-packages/numpy/core/include -I/usr/include/python3.6m -c pyk4a/pyk4a.cpp -o build/temp.linux-aarch64-3.6/pyk4a/pyk4a.o
  In file included from /usr/lib/python3/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1809:0,
                   from /usr/lib/python3/dist-packages/numpy/core/include/numpy/ndarrayobject.h:18,
                   from /usr/lib/python3/dist-packages/numpy/core/include/numpy/arrayobject.h:4,
                   from pyk4a/pyk4a.cpp:2:
  /usr/lib/python3/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
   #warning "Using deprecated NumPy API, disable it by " \
    ^~~~~~~
  pyk4a/pyk4a.cpp:4:10: fatal error: k4a/k4a.h: No such file or directory
   #include <k4a/k4a.h>
            ^~~~~~~~~~~
  compilation terminated.
  error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for pyk4a
  Running setup.py clean for pyk4a
Failed to build pyk4a
Installing collected packages: pyk4a
  Running setup.py install for pyk4a ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-2pf4lz14/pyk4a/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-t0682us4-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-aarch64-3.6
    creating build/lib.linux-aarch64-3.6/pyk4a
    copying pyk4a/pyk4a.py -> build/lib.linux-aarch64-3.6/pyk4a
    copying pyk4a/config.py -> build/lib.linux-aarch64-3.6/pyk4a
    copying pyk4a/__init__.py -> build/lib.linux-aarch64-3.6/pyk4a
    running build_ext
    building 'k4a_module' extension
    creating build/temp.linux-aarch64-3.6
    creating build/temp.linux-aarch64-3.6/pyk4a
    aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/lib/python3/dist-packages/numpy/core/include -I/usr/include/python3.6m -c pyk4a/pyk4a.cpp -o build/temp.linux-aarch64-3.6/pyk4a/pyk4a.o
    In file included from /usr/lib/python3/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1809:0,
                     from /usr/lib/python3/dist-packages/numpy/core/include/numpy/ndarrayobject.h:18,
                     from /usr/lib/python3/dist-packages/numpy/core/include/numpy/arrayobject.h:4,
                     from pyk4a/pyk4a.cpp:2:
    /usr/lib/python3/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
     #warning "Using deprecated NumPy API, disable it by " \
      ^~~~~~~
    pyk4a/pyk4a.cpp:4:10: fatal error: k4a/k4a.h: No such file or directory
     #include <k4a/k4a.h>
              ^~~~~~~~~~~
    compilation terminated.
    error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-2pf4lz14/pyk4a/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-t0682us4-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-2pf4lz14/pyk4a/

lpasselin, 2020-05-05:
Hi, you probably forgot to add the k4a lib to the path.
Check the readme for specific windows instructions.
Can you show the command you used to install with pip?

vnt1537, 2020-05-06:
Can you please tell me to which path do we need to add this k4alib? I added it to the folder where my build for k4aviewer is.
 I tried using 
sudo pip install pyk4a
sudo pip3 install pyk4a 
And even sudo -H pip3 install pyk4a

lpasselin, 2020-05-07:
pip3 usually refers to pip on recent systems. python 2 is not supported for this library.

Please read the readme. You need to specify the path to the pip command. 

I will close as this is a problem with your management of your windows environment.
I will continue to answer your questions in this thread.

closed 2020-05-07 by lpasselin

vnt1537, 2020-05-07:
Hey I am running it on Nvidia Xavier. Can you please specify from where can I download this k4a.lib? And where should I place it? I am sorry for such basic doubts but I really want to use this python wrapper. I installed the SDK via this link
 https://github.com/valdivj/Azure-for-Kinect-Jetson-nano/tree/10a564c117f3ccab37bdb0ada4fd6910c289b6d3

vnt1537, 2020-05-07:
i think the error is somewhere here

 pyk4a/pyk4a.cpp:4:10: fatal error: k4a/k4a.h: No such file or directory
     #include <k4a/k4a.h>
              ^~~~~~~~~~~
    compilation terminated.
    error: command 'gcc' failed with exit status 1

vnt1537, 2020-05-07:
The issue is solved thank you.

kramakr4, 2020-08-19:
Hi! I am facing a similar issue with pyk4a installation. Here is what I use:

**C:\Program Files\Azure Kinect SDK v1.4.1\tools> pip install pyk4a**
Collecting pyk4a
  Using cached pyk4a-0.3.tar.gz (6.9 kB)
Requirement already satisfied: numpy in c:\users\dexterity\appdata\local\programs\python\python38-32\lib\site-packages (from pyk4a) (1.19.1)
Using legacy 'setup.py install' for pyk4a, since package 'wheel' is not installed.
Installing collected packages: pyk4a
    Running setup.py install for pyk4a ... error
    ERROR: Command errored out with exit status 1:
     command: 'c:\users\dexterity\appdata\local\programs\python\python38-32\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\dexterity\\AppData\\Local\\Temp\\pip-install-zeusr1m1\\pyk4a\\setup.py'"'"'; __file__='"'"'C:\\Users\\dexterity\\AppData\\Local\\Temp\\pip-install-zeusr1m1\\pyk4a\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\dexterity\AppData\Local\Temp\pip-record-zoku34ab\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\dexterity\appdata\local\programs\python\python38-32\Include\pyk4a'
         cwd: C:\Users\dexterity\AppData\Local\Temp\pip-install-zeusr1m1\pyk4a\
    Complete output (19 lines):
    running install
    running build
    running build_py
    creating build
    creating build\lib.win32-3.8
    creating build\lib.win32-3.8\pyk4a
    copying pyk4a\config.py -> build\lib.win32-3.8\pyk4a
    copying pyk4a\pyk4a.py -> build\lib.win32-3.8\pyk4a
    copying pyk4a\__init__.py -> build\lib.win32-3.8\pyk4a
    running build_ext
    building 'k4a_module' extension
    creating build\temp.win32-3.8
    creating build\temp.win32-3.8\Release
    creating build\temp.win32-3.8\Release\pyk4a
    C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ic:\users\dexterity\appdata\local\programs\python\python38-32\lib\site-packages\numpy\core\include -Ic:\users\dexterity\appdata\local\programs\python\python38-32\include -Ic:\users\dexterity\appdata\local\programs\python\python38-32\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /EHsc /Tppyk4a/pyk4a.cpp /Fobuild\temp.win32-3.8\Release\pyk4a/pyk4a.obj
    pyk4a.cpp
    c:\users\dexterity\appdata\local\programs\python\python38-32\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(14) : Warning Msg: Using deprecated NumPy API, disable it with #define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
    pyk4a/pyk4a.cpp(4): fatal error C1083: Cannot open include file: 'k4a/k4a.h': No such file or directory
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.27.29110\\bin\\HostX86\\x86\\cl.exe' failed with exit status 2
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'c:\users\dexterity\appdata\local\programs\python\python38-32\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\dexterity\\AppData\\Local\\Temp\\pip-install-zeusr1m1\\pyk4a\\setup.py'"'"'; __file__='"'"'C:\\Users\\dexterity\\AppData\\Local\\Temp\\pip-install-zeusr1m1\\pyk4a\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\dexterity\AppData\Local\Temp\pip-record-zoku34ab\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\dexterity\appdata\local\programs\python\python38-32\Include\pyk4a' Check the logs for full command output.

I have also tried using this path:
C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\bin>

And also this command: 
pip install pyk4a --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.2.0\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.2.0\sdk\windows-desktop\amd64\release\lib"

Any combination of the above two paths and commands still gives similar error. Could you please help me with this? I am sorry if my question is too naive or dumb. I am new to coding and kinect environments.

kramakr4, 2020-08-19:
Hi! I am facing a similar issue with pyk4a installation. Here is what I use:

C:\Program Files\Azure Kinect SDK v1.4.1\tools> pip install pyk4a
Collecting pyk4a
Using cached pyk4a-0.3.tar.gz (6.9 kB)
Requirement already satisfied: numpy in c:\users\krr\appdata\local\programs\python\python38-32\lib\site-packages (from pyk4a) (1.19.1)
Using legacy 'setup.py install' for pyk4a, since package 'wheel' is not installed.
Installing collected packages: pyk4a
Running setup.py install for pyk4a ... error
ERROR: Command errored out with exit status 1:
command: 'c:\users\krr\appdata\local\programs\python\python38-32\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\Users\krr\AppData\Local\Temp\pip-install-zeusr1m1\pyk4a\setup.py'"'"'; file='"'"'C:\Users\krr\AppData\Local\Temp\pip-install-zeusr1m1\pyk4a\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(file);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, file, '"'"'exec'"'"'))' install --record 'C:\Users\krr\AppData\Local\Temp\pip-record-zoku34ab\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\krr\appdata\local\programs\python\python38-32\Include\pyk4a'
cwd: C:\Users\krr\AppData\Local\Temp\pip-install-zeusr1m1\pyk4a
Complete output (19 lines):
running install
running build
running build_py
creating build
creating build\lib.win32-3.8
creating build\lib.win32-3.8\pyk4a
copying pyk4a\config.py -> build\lib.win32-3.8\pyk4a
copying pyk4a\pyk4a.py -> build\lib.win32-3.8\pyk4a
copying pyk4a_init_.py -> build\lib.win32-3.8\pyk4a
running build_ext
building 'k4a_module' extension
creating build\temp.win32-3.8
creating build\temp.win32-3.8\Release
creating build\temp.win32-3.8\Release\pyk4a
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ic:\users\krr\appdata\local\programs\python\python38-32\lib\site-packages\numpy\core\include -Ic:\users\krr\appdata\local\programs\python\python38-32\include -Ic:\users\krr\appdata\local\programs\python\python38-32\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /EHsc /Tppyk4a/pyk4a.cpp /Fobuild\temp.win32-3.8\Release\pyk4a/pyk4a.obj
pyk4a.cpp
c:\users\krr\appdata\local\programs\python\python38-32\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(14) : Warning Msg: Using deprecated NumPy API, disable it with #define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
pyk4a/pyk4a.cpp(4): fatal error C1083: Cannot open include file: 'k4a/k4a.h': No such file or directory
error: command 'C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\HostX86\x86\cl.exe' failed with exit status 2
----------------------------------------
ERROR: Command errored out with exit status 1: 'c:\users\krr\appdata\local\programs\python\python38-32\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\Users\krr\AppData\Local\Temp\pip-install-zeusr1m1\pyk4a\setup.py'"'"'; file='"'"'C:\Users\krr\AppData\Local\Temp\pip-install-zeusr1m1\pyk4a\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(file);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, file, '"'"'exec'"'"'))' install --record 'C:\Users\krr\AppData\Local\Temp\pip-record-zoku34ab\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\krr\appdata\local\programs\python\python38-32\Include\pyk4a' Check the logs for full command output.

I have also tried using this path:
C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\bin>

And also this command:
pip install pyk4a --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.2.0\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.2.0\sdk\windows-desktop\amd64\release\lib"

Any combination of the above two paths and commands still gives similar error. Could you please help me with this? I am sorry if my question is too naive or dumb. I am new to coding and kinect environments.

lpasselin, 2020-08-20:
`fatal error C1083: Cannot open include file: 'k4a/k4a.h': No such file or directory` is the error. You need to specify the path to k4a.h file by telling the compiler where they are. Usually compilers have a `-I` option to specify directories to include.

`pip install pyk4a` this will not work unless the includes (`-I`) and libraries (`-L`) are somehow given to the compiler. Which is why we need to specify their paths by adding options with `--global-option`.

For the second command you used, I guess you didn't notice you were including directories of a k4a SDK **v1.2.0** while you seem to have v1.4.1 installed. 

Try this instead:
`pip install pyk4a --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"`

kramakr4, 2020-08-20:
Thanks a lot for getting back to me! I did make the change to the sdk version mentioned in the command to the one in my PC. The includes and libraries path mentioned do exist. However, when I try the above command I am getting the following error:

c:\users\krr\appdata\local\programs\python\python38-32\lib\site-packages\pip\_internal\commands\install.py:236: UserWarning: Disabling all use of wheels due to the use of --build-option / --global-option / --install-option.
  cmdoptions.check_install_build_global(options)
Collecting pyk4a
  Using cached pyk4a-0.3.tar.gz (6.9 kB)
Requirement already satisfied: numpy in c:\users\krr\appdata\local\programs\python\python38-32\lib\site-packages (from pyk4a) (1.19.1)
Skipping wheel build for pyk4a, due to binaries being disabled for it.
Installing collected packages: pyk4a
    Running setup.py install for pyk4a ... error
    ERROR: Command errored out with exit status 1:
     command: 'c:\users\krr\appdata\local\programs\python\python38-32\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\krr\\AppData\\Local\\Temp\\pip-install-821jmzpq\\pyk4a\\setup.py'"'"'; __file__='"'"'C:\\Users\\krr\\AppData\\Local\\Temp\\pip-install-821jmzpq\\pyk4a\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' build_ext '-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include' '-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib' install --record 'C:\Users\krr\AppData\Local\Temp\pip-record-oeoexwqt\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\krr\appdata\local\programs\python\python38-32\Include\pyk4a'
         cwd: C:\Users\krr\AppData\Local\Temp\pip-install-821jmzpq\pyk4a\
    Complete output (36 lines):
    running build_ext
    building 'k4a_module' extension
    creating build
    creating build\temp.win32-3.8
    creating build\temp.win32-3.8\Release
    creating build\temp.win32-3.8\Release\pyk4a
    C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\HostX86\x86\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ic:\users\krr\appdata\local\programs\python\python38-32\lib\site-packages\numpy\core\include "-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" -Ic:\users\krr\appdata\local\programs\python\python38-32\include -Ic:\users\krr\appdata\local\programs\python\python38-32\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /EHsc /Tppyk4a/pyk4a.cpp /Fobuild\temp.win32-3.8\Release\pyk4a/pyk4a.obj
    pyk4a.cpp
    c:\users\krr\appdata\local\programs\python\python38-32\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(14) : Warning Msg: Using deprecated NumPy API, disable it with #define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
    creating C:\Users\krr\AppData\Local\Temp\pip-install-821jmzpq\pyk4a\build\lib.win32-3.8
    C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\HostX86\x86\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO "/LIBPATH:C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib" /LIBPATH:c:\users\krr\appdata\local\programs\python\python38-32\libs /LIBPATH:c:\users\krr\appdata\local\programs\python\python38-32\PCbuild\win32 "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\ATLMFC\lib\x86" "/LIBPATH:C:\Program       Creating library build\temp.win32-3.8\Release\pyk4a\k4a_module.cp38-win32.lib and object build\temp.win32-3.8\Release\pyk4a\k4a_module.cp38-win32.exp
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_image_release
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_capture_get_depth_image
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_image_get_buffer
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_transformation_create
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_device_get_calibration
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_device_set_color_control
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_capture_release
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_device_stop_cameras
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_transformation_destroy
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_image_create
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_image_get_format
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_device_get_capture
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_device_open
    pyk4a.obj : error LNK2001: unresolved external symbol __imp__k4a_capture_create
[... cut ...] "'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' build_ext '-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include' '-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib' install --record 'C:\Users\krr\AppData\Local\Temp\pip-record-oeoexwqt\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\krr\appdata\local\programs\python\python38-32\Include\pyk4a' Check the logs for full command output.

kramakr4, 2020-08-20:
Thank you very much! This worked!

kramakr4, 2020-08-20:
When I run the following command, I get an error:

**from pyk4a import PyK4A**
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\krr\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\pyk4a\__init__.py", line 2, in <module>
    from .pyk4a import *
  File "C:\Users\krr\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\pyk4a\pyk4a.py", line 2, in <module>
    import k4a_module
ImportError: DLL load failed while importing k4a_module: The specified module could not be found.

zapaishchykova, 2021-01-25:
Hi @kramakr4! did you resolve this issue? I have all of the paths in the env of windows, and after the install was successful i also get
```
import k4a_module
ImportError: DLL load failed while importing k4a_module: The specified module could not be found.
```
[image]

Do you have any suggestions?

lpasselin, 2021-04-09:
@zapaishchykova sorry I never saw this. Did you manage to make it work?

zapaishchykova, 2021-04-09:
@lpasselin oh yes, a bit of manipulation with paths and it worked!

dmorris0, 2021-05-16:
@zapaishchykova What was the trick?  I'm getting the same problem, and can't seem to find the right path to add.

zapaishchykova, 2021-05-17:
@dmorris0 
[image]

dmorris0, 2021-05-17:
Thanks -- but unfortunately adding that line to my path and restarting my terminal doesn't work for me.  Still get the error:
```
>>> from pyk4a import PyK4A
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\morri\Source\envs\cv\lib\site-packages\pyk4a\__init__.py", line 1, in <module>
    import k4a_module
ImportError: DLL load failed while importing k4a_module: The specified module could not be found.
```

zapaishchykova, 2021-05-17:
@dmorris0 can you show me the contents of your \release\bin?

dmorris0, 2021-05-17:
Here is: C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\bin
[image]

zapaishchykova, 2021-05-17:
Do you specify in python path of the module? 
```python
import sys
sys.path.insert(1, '../pyKinectAzure/')
import numpy as np
from pyKinectAzure import pyKinectAzure,_k4a, postProcessing
import cv2
import time
import os

paths = []

# Path to the module
modulePath = r'C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\bin'

if __name__ == "__main__":
	# Initialize the library with the path containing the module
	pyK4A = pyKinectAzure(modulePath)

	# Open device
	num_connected_devices = pyK4A.device_get_installed_count()
	print("[LOG]: Number of connected devices ",num_connected_devices)
```

dmorris0, 2021-05-17:
Ah interesting.  I am trying to find out where you `pyKinectAzure` folder came from.  That's not in the `pyk4a` repo as far as I can tell nor in the Azure Kinect SDK.

zapaishchykova, 2021-05-17:
It should be this one 
https://github.com/ibaiGorordo/pyKinectAzure

shagren, 2021-05-18:
@dmorris0 , please check this: https://github.com/etiennedub/pyk4a/issues/117#issuecomment-816801446

dmorris0, 2021-05-19:
@shagren that looks exactly like what I need, but I run it and I get the same module error.

juancarlosmiranda, 2021-07-18:
I had the same problem, but I solved applying #117
The solution given by @shagren was very useful.
 The steps applied were:
* pip install pyk4a --no-use-pep517 --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"
* Path checked in: C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\bin
* A helper functions in a file

##############################
#helper_path.py
##############################
import sys
import os
from path import Path

def _add_dll_directory(path: Path):
    from ctypes import c_wchar_p, windll  # type: ignore
    from ctypes.wintypes import DWORD
    AddDllDirectory = windll.kernel32.AddDllDirectory
    AddDllDirectory.restype = DWORD
    AddDllDirectory.argtypes = [c_wchar_p]
    AddDllDirectory(str(path))

def kinect():
    if sys.platform != "win32":
        return
    env_path = os.getenv("KINECT_LIBS", None)
    if env_path:
        candidate = Path(env_path)
        dll = candidate / "k4a.dll"
        if dll.exists():
            _add_dll_directory(candidate)
            return
    #autodetecting
    program_files = Path("C:\\Program Files\\")
    for dir in sorted(program_files.glob("Azure Kinect SDK v*"), reverse=True):
        candidate = dir / "sdk" / "windows-desktop" / "amd64" / "release" / "bin"
        dll = candidate / "k4a.dll"
        if dll.exists():
            _add_dll_directory(candidate)
            return

#---- end of helper_path.py----

#############################
#example.py
#############################
import helper_path as hp

hp.kinect()
from pyk4a import PyK4A
from matplotlib import pyplot as plt

#Load camera with the default config
k4a = PyK4A()
k4a.start()

#Get the next capture (blocking function)
capture = k4a.get_capture()
img_color = capture.color
img_depth = capture.depth
img_ir = capture.ir

#create figure
fig = plt.figure(1)
plt.subplot(131)
plt.imshow(img_depth)
plt.title("DEPTH")

plt.subplot(132)
plt.imshow(img_ir)
plt.title("IR")

#Display with pyplot
plt.subplot(133)
plt.imshow(img_color[:, :, 2::-1]) # BGRA to RGB
plt.show()

#---- end of example.py----

lpasselin, 2021-07-18:
This problem is fixed in the dev.
The next version will solve these windows install issues.

juancarlosmiranda, 2021-07-18:
Thanks @lpasselin !!! my respects and congratulations for the great work!!

lpasselin, 2021-07-18:
Thank @shagren!

YoniChechik, 2022-03-06:
In windows: adding `C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\bin` to the PATH env var (via `edit envinronment variables`) solved it for me