etiennedub/pyk4a #30: Joints positions access

robinfays12, 2020-06-02:
Hello,

Thanks for this helpfull code !!

I have a question. Do you know how can we access the joints positions of the skeleton ?

Thanks a lot,

Robin Fays

lpasselin, 2020-06-02:
Checkout this wip merge request. It was working but we aren't sure how to integrate it yet.
https://github.com/etiennedub/pyk4a/pull/11

robinfays12, 2020-06-03:
Thanks a lot !

It says I don't have the attribute 'device_get_pose_data'. Do you know how I can add it ?

[image]

lpasselin, 2020-06-03:
I realized the up to date version of the body tracking branch was not merged properly. The body tracking functions are gone. Last working version was at this commit: https://github.com/etiennedub/pyk4a/pull/11/commits/049244356c37c8d6593d0cba7a8a878b7b185afa so you should install this version. Sorry about that.

Also you need to add a specific command with pip to enable body tracking. See the readme of the pull request for the command.

lpasselin, 2020-06-03:
Sorry I don't have access to my kinect azure. I'll fix this mess when I do.

robinfays12, 2020-06-04:
Ho, okay don't worry. Thank you for your answer. I'm trying to install the body tracking but it said it is not recognized. Did you add something to your environment variables ?

pip install pyk4a --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.0\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.0\sdk\windows-desktop\amd64\release\lib" --global-option="build_ext" --global-option="--enable-body-tracking"

robinfays12, 2020-06-04:
I mean this installation of pyk4a is working well without the extension :

--global-option="build_ext" --global-option="--enable-body-tracking"

lpasselin, 2020-06-04:
> I mean the installation of pyk4a is working well without the extension :

I really don't know. If you're using the version at the commit previously
mentioned it should be working.

I should be able to test next week.

robinfays12, 2020-06-05:
Ok, it would be nice if you can have a look. The depth is working well, but not the joints.

lpasselin, 2020-06-11:
Sorry I still don't have access to my k4a. 

The joints are not working with the specific commit? You should try installing it with `pip install git+git://github.com/lpasselin/pyk4a.git@049244356c37c8d6593d0cba7a8a878b7b185afa --upgrade --global-option="build_ext" --global-option="--enable-body-tracking"`

lpasselin, 2020-06-11:
Let's fix the body tracking issues in the related pull request https://github.com/etiennedub/pyk4a/pull/11

closed 2020-06-11 by lpasselin