etiennedub/pyk4a #31: Missing functions in k4a_module

tagelokoy, 2020-06-10:
Hi,
Thanks for making this repo, it looks like a promising solution for running the Kinect through Python. I have some issues after installing it though, not all the functions from the cpp source file end up in the k4a_module, like `device_get_depth_image()` or `transformation_depth_image_to_color_camera()`. I have tried uninstalling and reinstalling a couple times, with the path to both SKD v.1.2.0 and v.1.4.0, as in your installation instruction. Do you know why this happens, or how I can have all the necessary functions in the module?
The functions I need the most are getting depth, RGB-D or point cloud, and calibration data for using OpenCV with aruco boards etc.

**Screenshot of k4a_module.py:**

[image]
[image]

lpasselin, 2020-06-11:
I am not sure how to generate the file you are seeing. It seems to be a summary of the pyk4a.cpp file (k4a_module).

`device_get_depth_image` is in your screenshot.

`transformation_depth_image_to_color_camera` is not there because it is not in pyk4a version 0.3 (current version on pypi). You installed with `pip install pyk4a [...]` and the code is not yet up to date with this repository yet so it is normal! The actual code is at this tag https://github.com/etiennedub/pyk4a/tree/0.3

You should still be able to use the library with version 0.3

If you want to use the master branch, you can still do it by installing from git with `pip install git+https://github.com/etiennedub/pyk4a.git --upgrade`

closed 2020-06-11 by lpasselin

tagelokoy, 2020-06-16:
Thanks for helping solve what was actually an installation problem on my part. Using `pip install git+https://github.com/etiennedub/pyk4a.git --upgrade` didn't work on its own, I had to had `--global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.0\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.0\sdk\windows-desktop\amd64\release\lib"`. 

Are you considering adding the latest version from master to pypi? I would like to add this wrapper as a dependency in another package, but it seems kinda tricky this way. Feel no obligation to answer this, the issue is solved and closed of course.

lpasselin, 2020-06-16:
Yeah we will probably release it soon but there are still some functions to test.
When the covid situation settles down I will have access to my kinect.