etiennedub/pyk4a #33: How do i get the depth camera intrinsic matrix?

Elody-07, 2020-07-14:
Thanks for your project! 

Is there any function i can use to get the camera intrinsic matrix?

oliverschlick, 2020-07-20:
Thanks for your project as well!

I also got a question... i want to capture the depth image with your python code and safe it to a new file. When i try it myelf the depth image is way to dark not like the image you get when you start the viewer. Maybe you can help me, i am a beginner ..
Thanks

lpasselin, 2020-07-20:
@oliverschlick The depth map is saved as a 16bit grayscale PNG. You must rescale to see anything. 8bit colors wound not fit the full range of the kinect azure depth sensor.

lpasselin, 2020-07-20:
@Elody-07 https://github.com/etiennedub/pyk4a/blob/887d89bedf1854a89d31ad4932399ca7c6f2393b/pyk4a/pyk4a.py#L46

closed 2020-07-20 by lpasselin

etiennedub, 2020-07-20:
@Elody-07 You can export the whole calibration as a json file with the method `Pyk4a.save_calibration_json`.
The file include all the calibration use by the device (gyro, cameras intrincs and camera extrinsic) and there is no documentation.
If you have difficulty to convert their intrinsic representation to something else (eg. OpenCV) I could help you.

@oliverschlick it's better to keep discussion related to the initial issue. If you have another issue feel free to open a new one.

oliverschlick, 2020-07-23:
Thank you for your quick response. First I cloned your repository to have the same files.  Now I got some issues with saving the file, you suggested me to use pyk4a.save_calibration_json. I see that this is a method you defined in your pyk4a class, but when i try to call it in your Viewer.py it doesnt work. I am working with pycharm where it should auto complete the function i try to use when i am typing, but it‘s not. I created an instance like img1 = PyK4A() and then tried to call the function with img1.save_calibration_json()… 
It looks like it’s not loading the package Right. The failure i get is PyK4A object has no Attribute save_calibration_json, it’s the same with other functions from the class PyK4A, although in the Code it says from pyk4a Import PyK4A.

The second Problem i have is that i tried to Change the Resolution in the config from 720p to 3072p I also changed the Frame rate therefore. I expected a Picture like when you run the camera with the installed sdk Viewer, but it just crops the Picture and zooms in.

Hopefully i explained it good enough for you to understand, it’s a bit difficult for me to explain the issues.
I really appreciate your help! Thanks and greetings from Germany.

Oliver Schlick

Von: etiennedub
Gesendet: Montag, 20. Juli 2020 22:44
An: etiennedub/pyk4a
Cc: oliverschlick; Mention
Betreff: Re: [etiennedub/pyk4a] How do i get the depth camera intrinsicmatrix? (#33)

@Elody-07 You can export the whole calibration as a json file with the method Pyk4a.save_calibration_json.
The file include all the calibration use by the device (gyro, cameras intrincs and camera extrinsic) and there is no documentation.
If you have difficulty to convert their intrinsic representation to something else (eg. OpenCV) I could help you.
@oliverschlick it's better to keep discussion related to the initial issue. If you have another issue feel free to open a new one.

etiennedub, 2020-07-26:
My guess is that you don't have the `save_calibration_json` method because it is not available from the PyPi repo yet (sorry I don't often update it) and you probably install the package from there (`pip install pyk4a`).
I suggest that you try installing the package from the github repo as follow:
```
pip uninstall pyk4a
pip install git+https://github.com/etiennedub/pyk4a
```
That should uninstall the current version and install the new one direcly from the github repo.

For the other problem with the resolution and frame rate, that's the expected behaviour. The kinect doesn't support 30fps for all resolution.

Elody-07, 2020-07-26:
> My guess is that you don't have the `save_calibration_json` method because it is not available from the PyPi repo yet (sorry I don't often update it) and you probably install the package from there [...]

I did install the package using pip. It's just that when I install from the github repo, some errors pop out. It seems like I did not have k4a.h or something. Finally I get the intrinsic using C++ SDK.  Thanks for your response anyway. Have a nice day!

FelipeDelgadoR, 2020-07-27:
Hi @etiennedub 
I'm currently using your repo to work with my Azure Kinect and it works really nice!
I'm having some issues with the intrinsics that I got from this function (save_calibration_json)
I have the current intrinsics that I got from the Open3D Method (with the mkv reader) and I got this (and they work):
1920, 1080, 919.42980957, 919.3795166, 953.07250977, 551.76287842
The problem now is that using your function I got this:
0.49665233492851257,
0.50851589441299438,
0.47886967658996582,
0.6384580135345459,
0.32674625515937805,
-2.3125674724578857,
1.3249306678771973,
0.21215163171291351,
-2.1464962959289551,
1.2583736181259155,
0,
0,
-0.00038163267890922725,
 0.000783831172157079

and none of these values look like the ones that I got
I think that this is some form of  problem converting these values and I'd really appreciate some help from you!
Thanks for everything

oliverschlick, 2020-07-28:
Okay first problem solved, now I got the right package loaded. I need jpg/jpeg files from the camera stream, is there a possibility to get those files, because I think with json files i can't really work? I think I should first explain, what i want to do. I am writing my bachelorthesis right now, the topic is "Detection of defects by means of camera monitoring in Fused Deposition Modeling" . Now I got the Micorsoft Azure Kinect camera from my institute I am working at. All my coding for now is written in Python and i would like to combine it with taking pictures via Python Code.
The second problem, maybe i didn't quite understand what you were saying, but I lowered the FPS-rate to 15fps or even 5fps to change the resolution to 2160p or higher. I expected to get a higher resoluted image, but it just zooms in and crops the picture. When i am changing the resolution in the azure kienct viewer application to 2160p and the frame rate to 15fps the image is the same but higher resoluted without zooming in or cropping the picture.

Really appreciate your help, thank you!

Von: etiennedub
Gesendet: Sonntag, 26. Juli 2020 03:32
An: etiennedub/pyk4a
Cc: oliverschlick; Mention
Betreff: Re: [etiennedub/pyk4a] How do i get the depth camera intrinsicmatrix? (#33)

My guess is that you don't have the save_calibration_json method because it is not available from the PyPi repo yet (sorry I don't often update it) and you probably install the package from there (pip install pyk4a).
I suggest that you try installing the package from the github repo as follow:
pip uninstall pyk4a
pip install git+https://github.com/etiennedub/pyk4a
That should uninstall the current version and install the new one direcly from the github repo.
For the other problem with the resolution and frame rate, that's the expected behaviour. The kinect doesn't support 30fps for all resolution.

lpasselin, 2020-07-28:
Are you talking about the depth image or the RGB image? The depth nimage
only has two configurations. You should read the k4a docs about that.

For RGB, can you try with a resolution of 3072p?

oliverschlick, 2020-07-28:
I am talking about the rgb image, I slightly changed the viewer.py by changing the line:

k4a= PyK4A(Config(color_resolution=ColorResolution.Res_720P,..., camera_fps=pyk4a. FPS. FPS_30,...)) to k4a= PyK4A(Config(color_resolution=ColorResolution.Res_3072P,..., camera_fps=pyk4a. FPS. FPS_5,...))

lpasselin, 2020-07-28:
I really don't know what could be the problem is the official k4a viewer is working properly.

Are you sure FPS and resolution are the only thing you changed from the viewer.py file?

Maybe try to unplug the kinect USB and power, plug again and retry with the updated config? Make sure the official viewer is off. 

My feeling is that somehow the resolution is modified in the official viewer while the resolution in pyk4a is still 720p... Not sure if this is possible.
Maybe there is a bug in pyk4a since the latest 1.4.1 update but I doubt that. I tested 3072p with 1.4.1 last friday but not with the viewer.py script.