etiennedub/pyk4a #34: Problems installing from github repo

FelipeDelgadoR, 2020-07-27:
Hi!
I'm currently trying to install pyk4a with the following method:

pip install git+https://github.com/etiennedub/pyk4a.git --upgrade --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.0\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.0\sdk\win
dows-desktop\amd64\release\lib"

and I got this error:

Installing collected packages: pyk4a
  Running setup.py install for pyk4a ... error
    Complete output from command C:\Users\felip\Rebartek\vision-system\Scripts\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\felip\\AppData\\Local\\Temp\\pip-req-build-kgv40y5d\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().r
eplace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" build_ext "-IC:\Program Files\Azure Kinect SDK v1.4.0\sdk\include" "-LC:\Program Files\Azure Kinect SDK v1.4.0\sdk\win" install --record C:\Users\felip\AppData\Local\Temp\pip-record-y7vyiq7t\install
-record.txt --single-version-externally-managed --compile --install-headers C:\Users\felip\Rebartek\vision-system\include\site\python3.7\pyk4a:
    running build_ext
    building 'k4a_module' extension
    creating build
    creating build\temp.win-amd64-3.7
    creating build\temp.win-amd64-3.7\Release
    creating build\temp.win-amd64-3.7\Release\pyk4a
    C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.26.28801\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MT -IC:\Users\felip\Rebartek\vision-system\lib\site-packages\numpy\core\include "-IC:\Program Files\Azure Kinect SDK v1
.4.0\sdk\include" -IC:\Users\felip\Rebartek\vision-system\include -IC:\RoboDK\Python37\include -IC:\RoboDK\Python37\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Windows Kits\10\i
nclude\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows
Kits\10\include\10.0.18362.0\cppwinrt" /EHsc /Tppyk4a/pyk4a.cpp /Fobuild\temp.win-amd64-3.7\Release\pyk4a/pyk4a.obj
    pyk4a.cpp
    C:\Users\felip\Rebartek\vision-system\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(14) : Warning Msg: Using deprecated NumPy API, disable it with #define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
    pyk4a/pyk4a.cpp(204): warning C4244: 'initializing': conversion from 'npy_intp' to 'int', possible loss of data
    pyk4a/pyk4a.cpp(205): warning C4244: 'initializing': conversion from 'npy_intp' to 'int', possible loss of data
    creating C:\Users\felip\AppData\Local\Temp\pip-req-build-kgv40y5d\build\lib.win-amd64-3.7
    C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.26.28801\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /nodefaultlib:libucrt.lib ucrt.lib /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO "/LIBPATH:C:\Program Files\Azure Kinect SDK
 v1.4.0\sdk\win" /LIBPATH:C:\Users\felip\Rebartek\vision-system\libs /LIBPATH:C:\RoboDK\Python37\libs /LIBPATH:C:\RoboDK\Python37 /LIBPATH:C:\Users\felip\Rebartek\vision-system\PCbuild\amd64 "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\T
ools\MSVC\14.26.28801\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.18362.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.18362.0\um\x64" k4a.lib /EXPORT:PyInit_k4a_module build\temp.win-amd64-3.7\Release\pyk4a/pyk4a.obj /OUT
:build\lib.win-amd64-3.7\k4a_module.cp37-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.7\Release\pyk4a\k4a_module.cp37-win_amd64.lib
    LINK : fatal error LNK1181: cannot open input file 'k4a.lib'
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.26.28801\\bin\\HostX86\\x64\\link.exe' failed with exit status 1181

    ----------------------------------------
Command "C:\Users\felip\Rebartek\vision-system\Scripts\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\felip\\AppData\\Local\\Temp\\pip-req-build-kgv40y5d\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.c
lose();exec(compile(code, __file__, 'exec'))" build_ext "-IC:\Program Files\Azure Kinect SDK v1.4.0\sdk\include" "-LC:\Program Files\Azure Kinect SDK v1.4.0\sdk\win" install --record C:\Users\felip\AppData\Local\Temp\pip-record-y7vyiq7t\install-record.txt --single-ver
sion-externally-managed --compile --install-headers C:\Users\felip\Rebartek\vision-system\include\site\python3.7\pyk4a" failed with error code 1 in C:\Users\felip\AppData\Local\Temp\pip-req-build-kgv40y5d\

lpasselin, 2020-07-27:
`fatal error LNK1181: cannot open input file 'k4a.lib'`

I do not have much time to help you debug this right now but it looks like the paths you are providing does not satisfy the linker for k4a.lib.

Can you make sure the paths given do contain the required k4a.lib?

FelipeDelgadoR, 2020-07-28:
I only have this on my SYSTEM PATH right now:

C:\Program Files\Azure Kinect SDK v1.4.0\sdk\windows-desktop\amd64\release\bin

FelipeDelgadoR, 2020-07-28:
Nevermind, for some reason it started working today :/

I used the command provided by you

pip install git+https://github.com/etiennedub/pyk4a.git --upgrade --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.0\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.0\sdk\windows-desktop\amd64\release\lib"

closed 2020-07-28 by FelipeDelgadoR