etiennedub/pyk4a #35: Transforming calibration parameters

FelipeDelgadoR, 2020-07-28:
Hi @etiennedub
I'm currently using your repo to work with my Azure Kinect and it works really nice!
I'm having some issues with the intrinsics that I got from this function (save_calibration_json)
I have the current intrinsics that I got from the Open3D Method (with the mkv reader) and I got this (and they work):

1920, 1080, 919.42980957, 919.3795166, 953.07250977, 551.76287842

The problem now is that using your function I got this:

0.49665233492851257,
0.50851589441299438,
0.47886967658996582,
0.6384580135345459,
0.32674625515937805,
-2.3125674724578857,
1.3249306678771973,
0.21215163171291351,
-2.1464962959289551,
1.2583736181259155,
0,
0,
-0.00038163267890922725,
0.000783831172157079

and none of these values look like the ones that I got
I think that this is some form of problem converting these values and I'd really appreciate some help from you!
Thanks for everything

lpasselin, 2020-07-28:
Maybe Open3D calibration config is not the same format as the raw calibration values obtained from the device?
As Etienne said, pyk4a outputs the calibration in a format compatible with opencv.

etiennedub, 2020-07-28:
@lpasselin They are not directly compatible with OpenCV but can be converted.

Those 15 values are the intrinsic define here: https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__calibration__intrinsic__parameters__t_1_1__param.html
All the distortion coefficients can be directly used in OpenCV but not the other prams (fx, fy, cx and cy). There are normalize and that's why they don't look the same. 
You can "unormalize" but it depends on your resolution and other stuff. You should be able to do it by looking at this [code](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/95f1d95f1f335b57a350a80a3a62e98e1ee4258d/src/transformation/mode_specific_calibration.c).
If it's still not clear from the code, you can always  ask me.

closed 2020-07-30 by etiennedub

FelipeDelgadoR, 2020-09-21:
Hi @etiennedub 
I was really busy with lots of things but now I'm ready to resume this issue...
I was looking at the code that you send me and I'm not quite sure how this help me...
I'm using 1080p color resolution and both NFOV_UNBINNED and WFOV_UNBINNED to work with my kinects
how should I transform these values?

johan12345, 2021-03-23:
FYI: I have opened a pull request adding a function for accessing the intrinsic parameters in an OpenCV-compatible format: #113.