etiennedub/pyk4a #46: about get calibration matrix of camera

keithwpa, 2020-08-16:
it did not work, when i use k4a.save_calibration_json function.

Exception has occurred: AttributeError
'PyK4A' object has no attribute 'save_calibration_json'

lpasselin, 2020-08-17:
Please make sure you are using the most recent version of pyk4a. Can you tell us which version you are using?

keithwpa, 2020-08-18:
Requirement already satisfied: pyk4a in c:\users\ifl\anaconda3\envs\centernet\lib\site-packages (0.3)
Requirement already satisfied: numpy in c:\users\ifl\anaconda3\envs\centernet\lib\site-packages (from pyk4a) (1.18.5)

lpasselin, 2020-08-18:
@etiennedub can you check this out?

lpasselin, 2020-08-19:
I'm pretty sure you are not using the most recent pypi version. Try reinstalling with the upgrade argument (`pip install --upgrade` ...)

closed 2020-08-24 by lpasselin

keithwpa, 2020-08-25:
i try to do this but failed.

Collecting git+https://github.com/etiennedub/pyk4a.git
  Cloning https://github.com/etiennedub/pyk4a.git to c:\users\ifl\appdata\local\temp\pip-req-build-tugeyz1v
  Running command git clone -q https://github.com/etiennedub/pyk4a.git 'C:\Users\IFL\AppData\Local\Temp\pip-req-build-tugeyz1v'
Requirement already satisfied (use --upgrade to upgrade): pyk4a==0.3 from git+https://github.com/etiennedub/pyk4a.git in c:\users\ifl\anaconda3\envs\centernet\lib\site-packages
Requirement already satisfied: numpy in c:\users\ifl\anaconda3\envs\centernet\lib\site-packages (from pyk4a==0.3) (1.18.5)
Building wheels for collected packages: pyk4a
  Building wheel for pyk4a (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'C:\Users\IFL\anaconda3\envs\CenterNet\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\IFL\\AppData\\Local\\Temp\\pip-req-build-tugeyz1v\\setup.py'"'"'; __file__='"'"'C:\\Users\\IFL\\AppData\\Local\\Temp\\pip-req-build-tugeyz1v\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\IFL\AppData\Local\Temp\pip-wheel-xddjzrgd'
       cwd: C:\Users\IFL\AppData\Local\Temp\pip-req-build-tugeyz1v\
  Complete output (19 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.6
  creating build\lib.win-amd64-3.6\pyk4a
  copying pyk4a\config.py -> build\lib.win-amd64-3.6\pyk4a
  copying pyk4a\pyk4a.py -> build\lib.win-amd64-3.6\pyk4a
  copying pyk4a\__init__.py -> build\lib.win-amd64-3.6\pyk4a
  running build_ext
  building 'k4a_module' extension
  creating build\temp.win-amd64-3.6
  creating build\temp.win-amd64-3.6\Release
  creating build\temp.win-amd64-3.6\Release\pyk4a
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Users\IFL\anaconda3\envs\CenterNet\lib\site-packages\numpy\core\include -IC:\Users\IFL\anaconda3\envs\CenterNet\include -IC:\Users\IFL\anaconda3\envs\CenterNet\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /EHsc /Tppyk4a/pyk4a.cpp /Fobuild\temp.win-amd64-3.6\Release\pyk4a/pyk4a.obj
  pyk4a.cpp
  C:\Users\IFL\anaconda3\envs\CenterNet\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(14) : Warning Msg: Using deprecated NumPy API, disable it with #define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
  pyk4a/pyk4a.cpp(4): fatal error C1083: Datei (Include) kann nicht ge”ffnet werden: "k4a/k4a.h": No such file or directory
  error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.26.28801\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2
  ----------------------------------------
  ERROR: Failed building wheel for pyk4a
  Running setup.py clean for pyk4a
Failed to build pyk4a

keithwpa, 2020-08-25:
i try it but failed.

Collecting git+https://github.com/etiennedub/pyk4a.git
  Cloning https://github.com/etiennedub/pyk4a.git to c:\users\ifl\appdata\local\temp\pip-req-build-eibf32wv
  Running command git clone -q https://github.com/etiennedub/pyk4a.git 'C:\Users\IFL\AppData\Local\Temp\pip-req-build-eibf32wv'
Requirement already satisfied: numpy in c:\users\ifl\anaconda3\envs\centernet\lib\site-packages (from pyk4a==0.3) (1.18.5)
Building wheels for collected packages: pyk4a
  Building wheel for pyk4a (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'C:\Users\IFL\anaconda3\envs\CenterNet\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\IFL\\AppData\\Local\\Temp\\pip-req-build-eibf32wv\\setup.py'"'"'; __file__='"'"'C:\\Users\\IFL\\AppData\\Local\\Temp\\pip-req-build-eibf32wv\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\IFL\AppData\Local\Temp\pip-wheel-7wo53eos'
       cwd: C:\Users\IFL\AppData\Local\Temp\pip-req-build-eibf32wv\
  Complete output (19 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.6
  creating build\lib.win-amd64-3.6\pyk4a
  copying pyk4a\config.py -> build\lib.win-amd64-3.6\pyk4a
  copying pyk4a\pyk4a.py -> build\lib.win-amd64-3.6\pyk4a
  copying pyk4a\__init__.py -> build\lib.win-amd64-3.6\pyk4a
  running build_ext
  building 'k4a_module' extension
  creating build\temp.win-amd64-3.6
  creating build\temp.win-amd64-3.6\Release
  creating build\temp.win-amd64-3.6\Release\pyk4a
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Users\IFL\anaconda3\envs\CenterNet\lib\site-packages\numpy\core\include -IC:\Users\IFL\anaconda3\envs\CenterNet\include -IC:\Users\IFL\anaconda3\envs\CenterNet\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /EHsc /Tppyk4a/pyk4a.cpp /Fobuild\temp.win-amd64-3.6\Release\pyk4a/pyk4a.obj
  pyk4a.cpp
  C:\Users\IFL\anaconda3\envs\CenterNet\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(14) : Warning Msg: Using deprecated NumPy API, disable it with #define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
  pyk4a/pyk4a.cpp(4): fatal error C1083: Datei (Include) kann nicht ge”ffnet werden: "k4a/k4a.h": No such file or directory
  error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.26.28801\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2
  ----------------------------------------
  ERROR: Failed building wheel for pyk4a
  Running setup.py clean for pyk4a
Failed to build pyk4a
Installing collected packages: pyk4a
    Running setup.py install for pyk4a ... error
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\IFL\anaconda3\envs\CenterNet\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\IFL\\AppData\\Local\\Temp\\pip-req-build-eibf32wv\\setup.py'"'"'; __file__='"'"'C:\\Users\\IFL\\AppData\\Local\\Temp\\pip-req-build-eibf32wv\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\IFL\AppData\Local\Temp\pip-record-44pqx1ys\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\IFL\anaconda3\envs\CenterNet\Include\pyk4a'
         cwd: C:\Users\IFL\AppData\Local\Temp\pip-req-build-eibf32wv\
    Complete output (19 lines):
    running install
    running build
    running build_py
    creating build
    creating build\lib.win-amd64-3.6
    creating build\lib.win-amd64-3.6\pyk4a
    copying pyk4a\config.py -> build\lib.win-amd64-3.6\pyk4a
    copying pyk4a\pyk4a.py -> build\lib.win-amd64-3.6\pyk4a
    copying pyk4a\__init__.py -> build\lib.win-amd64-3.6\pyk4a
    running build_ext
    building 'k4a_module' extension
    creating build\temp.win-amd64-3.6
    creating build\temp.win-amd64-3.6\Release
    creating build\temp.win-amd64-3.6\Release\pyk4a
    C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Users\IFL\anaconda3\envs\CenterNet\lib\site-packages\numpy\core\include -IC:\Users\IFL\anaconda3\envs\CenterNet\include -IC:\Users\IFL\anaconda3\envs\CenterNet\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ [... cut ...] ; __file__='"'"'C:\\Users\\IFL\\AppData\\Local\\Temp\\pip-req-build-eibf32wv\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\IFL\AppData\Local\Temp\pip-record-44pqx1ys\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\IFL\anaconda3\envs\CenterNet\Include\pyk4a' Check the logs for full command output.