etiennedub/pyk4a #60: Loading DLL fails in Python 3.8.5

matnilswe, 2020-09-08:
Python 3.8.5 (with venv)
when importing: 
from pyk4a import PyK4A
I hit the issue:
"DLL load failed while importing k4a_module: The specified module could not be found"

DLL loading works when installing and importing from Python 3.7.5

lpasselin, 2020-09-08:
First time I hear this. Not sure how to help you.

Is python 3.7.5 also with venv?

Is it possible that the venv with python 3.8 tries to reuse the 3.7.5 version? Maybe try uninstalling and reinstalling using `--upgrade` ?

Can you try installing from the develop branch?

matnilswe, 2020-09-08:
The 3.7.5 Python I tried was in a conda environment. To be clear this is what I tried after failing with the Python 3.8.5 with venv. Don't think that the issue has anything to do with "reuse".  Hard to see exactly what fails since it's in "k4a_module.cp38-win_amd64.pyd" I guess.

lpasselin, 2020-09-09:
Which pip version are you using?

matnilswe, 2020-09-10:
pip 20.2.2

matnilswe, 2020-09-10:
FYI. upgraded to pip 20.2.3. 
Issue remains.

lpasselin, 2020-09-12:
I'm really not a windows expert.

`DLL load failed while importing k4a_module: The specified module could not be found` Is this related to the depthengine DLL?

There is a similar issue here https://github.com/etiennedub/pyk4a/issues/14

lpasselin, 2020-09-12:
@shagren do you know how to fix this? multiple users have the problem.

shagren, 2020-09-21:
@matnilswe , can you check this version: https://github.com/etiennedub/pyk4a/pull/68 ?

lpasselin, 2020-09-29:
Might be solved in master with https://github.com/etiennedub/pyk4a/pull/68
Would you mind trying to install with the command in the README?

matnilswe, 2020-09-29:
Used the install command you suggested. pip install pyk4a --no-use-pep517 --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.0\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.0\sdk\windows-desktop\amd64\release\lib"

Import seems to work now. Thanks! :-)

closed 2020-09-29 by lpasselin

matnilswe, 2020-10-12:
Eh, I must have tried it in my anaconda env. Seems to work on 3.8.2 anaconda, but now I seem to still have the same issue in my venv for 3.8.5. Sorry about the confusion.

ozgucbertug, 2020-11-14:
This issue still persists. Building the library works fine however when I try to run the example code on Windows, I receive ImportError: DLL load failed while importing k4a_module: The specified module could not be found.