etiennedub/pyk4a #69: Transforming calibration parameters (intrinsics)

FelipeDelgadoR, 2020-09-24:
Hi @etiennedub 
I've got these values from the RGB camera:

"ModelParameters": [
            0.49965196847915649,
            0.50621724128723145,
            0.47322136163711548,
            0.63089394569396973,
            0.44268873333930969,
            -2.6038174629211426,
            1.5527737140655518,
            0.32462942600250244,
            -2.4303467273712158,
            1.4789470434188843,
            0,
            0,
            -0.00035283574834465981,
            0.00099612073972821236
          ]

I know that the first 2 are cx, cy, fx and fy and that they are normalized so I can't use them inside opencv as the intrinsic matrix...
Right now i'm using this configuration:
K4A_FRAMES_PER_SECOND_15
K4A_COLOR_RESOLUTION_1080P
K4A_DEPTH_MODE_WFOV_UNBINNED

You gave me [this code](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/95f1d95f1f335b57a350a80a3a62e98e1ee4258d/src/transformation/mode_specific_calibration.c) to check it out, but I'm not sure if I understand it correctly
I'd appreciate if you give me some help on transforming these values, thanks again!

shagren, 2020-09-24:
https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__calibration__intrinsic__parameters__t_1_1__param.html

Like as all except MetricRadius are here.
You can find last in calibration json: https://github.com/etiennedub/pyk4a/blob/37b7305f65f6fd671e752d91ff4f803fedd1ff1e/tests/assets/calibration.json#L27

FelipeDelgadoR, 2020-09-25:
I found the function that defines the structure of the code:

typedef struct _k4a_camera_calibration_mode_info_t
{
    unsigned int calibration_image_binned_resolution[2];
    int crop_offset[2];
    unsigned int output_image_resolution[2];
} k4a_camera_calibration_mode_info_t;

so basically when we use the 1080p resolution we got these values:

case K4A_COLOR_RESOLUTION_1080P:
{
   k4a_camera_calibration_mode_info_t mode_info = { { 1920, 1440 }, { 0, 180 }, { 1920, 1080 } };

so basically according to this code we have to perform this operation:

float cx = params->param.cx * mode_info->calibration_image_binned_resolution[0];
    float cy = params->param.cy * mode_info->calibration_image_binned_resolution[1];
    float fx = params->param.fx * mode_info->calibration_image_binned_resolution[0];
    float fy = params->param.fy * mode_info->calibration_image_binned_resolution[1];

    cx -= mode_info->crop_offset[0];
    cy -= mode_info->crop_offset[1];

And I got these values:

cx = 959.3317794799805
cy = 548.9528274536133
fx = 908.5850143432617
fy = 908.4872817993164

The values that we got from this "recording procedure" that open3d offers are:

cx = 958.831787109375
cy = 548.45281982421875
fx = 908.58502197265625
fy = 908.4873046875

So I think that I got it now...
but now I've stumbled upon another problem.
In the calibration json file I got these values (on the rgb camera):

"Rt": {
          "Rotation": [
            0.99998795986175537,
            0.0047808918170630932,
            -0.0010995579650625587,
            -0.004640678409487009,
            0.99457055330276489,
            0.10396119207143784,
            0.0015906151384115219,
            -0.10395484417676926,
            0.99458074569702148
          ],
          "Translation": [
            -0.032096963375806808,
            -0.0019254388753324747,
            0.0040484610944986343
          ]
        },

According to me, this means the transformation matrix between the RGB camera and the DEPTH camera, is this right? what units are these? what is the structure of the matrix?

thanks in advance!

etiennedub, 2020-09-26:
Hi @FelipeDelgadoR, I think you got it right for the intrinsics paramaters and as you said, it really depends on the resolution. 

For the extrinsics parameters, those are not normalized and can be used directly in a 4x3 matrix. There is more information in the [OpenCV documentation.](https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html)
The form of the matrix is the following:
R[0], R[1], R[2], T[0],
R[3], R[4], R[5], T[1],
R[6], R[7], R[8], T[2]

Yes those the extrinsics parameters between both camera, the translation unit is in meter and for the rotation there is no unit because those are sin and cos values of the rotation by axis.

I hope it answer your question!

FelipeDelgadoR, 2020-09-28:
I was just thinking that after I posted the question, of course the T values are in meters!
Thanks @etiennedub! I'll close this issue...

closed 2020-09-28 by FelipeDelgadoR

ZdenekM, 2020-11-19:
How can I get `mode_info` using `pyk4a`? I searched the code but could not find it... Many thanks.

Btw, it would be great to have a utility function to convert intrinsic parameters to the OpenCV format. I think more people would appreciate it :-)

johan12345, 2021-03-23:
@ZdenekM 
> Btw, it would be great to have a utility function to convert intrinsic parameters to the OpenCV format. I think more people would appreciate it :-)

FYI: I have opened a pull request adding such a function for accessing the intrinsic parameters in an OpenCV-compatible format: #113.