etiennedub/pyk4a #70: pip install failed

ZhangLanTao, 2020-09-27:
I tried to install pyk4a in Win10 python37 using: 
pip install pyk4a --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.0\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.0\sdk\windows-desktop\amd64\release\lib"
but I get error like:
pyk4a/pyk4a.cpp(4): fatal error C1083: 无法打开包括文件: “k4a/k4a.h”: No such file or directory
Could you please help me to fix it, thanks very much

full output:
C:\Users\49878\AppData\Roaming\Python\Python37\site-packages\pip\_internal\commands\install.py:235: UserWarning: Disabling all use of wheels due to the use of --build-option / --global-option / --install-option.
  cmdoptions.check_install_build_global(options)
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: http://mirrors.aliyun.com/pypi/simple/
Collecting pyk4a
  Downloading http://mirrors.aliyun.com/pypi/packages/37/81/b0a364e78ffc0f7587f1d3085ae8fc0f9138ab36921b98478e86f3a18259/pyk4a-1.0.tar.gz (15 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Requirement already satisfied: numpy in c:\users\49878\appdata\roaming\python\python37\site-packages (from pyk4a) (1.16.2)
Building wheels for collected packages: pyk4a
  Building wheel for pyk4a (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: 'c:\program files\python37\python.exe' 'C:\Users\49878\AppData\Roaming\Python\Python37\site-packages\pip\_vendor\pep517\_in_process.py' build_wheel 'C:\Users\49878\AppData\Local\Temp\tmp17c3n2xf'
       cwd: C:\Users\49878\AppData\Local\Temp\pip-install-xflwx8cs\pyk4a
  Complete output (26 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.7
  creating build\lib.win-amd64-3.7\pyk4a
  copying pyk4a\calibration.py -> build\lib.win-amd64-3.7\pyk4a
  copying pyk4a\capture.py -> build\lib.win-amd64-3.7\pyk4a
  copying pyk4a\config.py -> build\lib.win-amd64-3.7\pyk4a
  copying pyk4a\errors.py -> build\lib.win-amd64-3.7\pyk4a
  copying pyk4a\playback.py -> build\lib.win-amd64-3.7\pyk4a
  copying pyk4a\pyk4a.py -> build\lib.win-amd64-3.7\pyk4a
  copying pyk4a\results.py -> build\lib.win-amd64-3.7\pyk4a
  copying pyk4a\transformation.py -> build\lib.win-amd64-3.7\pyk4a
  copying pyk4a\__init__.py -> build\lib.win-amd64-3.7\pyk4a
  copying pyk4a\py.typed -> build\lib.win-amd64-3.7\pyk4a
  running build_ext
  building 'k4a_module' extension
  creating build\temp.win-amd64-3.7
  creating build\temp.win-amd64-3.7\Release
  creating build\temp.win-amd64-3.7\Release\pyk4a
  C:\Program Files\VisualStudio2019\VC\Tools\MSVC\14.25.28610\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Users\49878\AppData\Local\Temp\pip-build-env-_ufdcbgq\overlay\Lib\site-packages\numpy\core\include "-Ic:\program files\python37\include" "-Ic:\program files\python37\include" "-IC:\Program Files\VisualStudio2019\VC\Tools\MSVC\14.25.28610\ATLMFC\include" "-IC:\Program Files\VisualStudio2019\VC\Tools\MSVC\14.25.28610\include" "-IC:\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Windows Kits\10\include\10.0.18362.0\cppwinrt" /EHsc /Tppyk4a/pyk4a.cpp /Fobuild\temp.win-amd64-3.7\Release\pyk4a/pyk4a.obj
  pyk4a.cpp
  C:\Users\49878\AppData\Local\Temp\pip-build-env-_ufdcbgq\overlay\Lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(14) : Warning Msg: Using deprecated NumPy API, disable it with #define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
  pyk4a/pyk4a.cpp(4): fatal error C1083: 无法打开包括文件: “k4a/k4a.h”: No such file or directory
  error: command 'C:\\Program Files\\VisualStudio2019\\VC\\Tools\\MSVC\\14.25.28610\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2
  ----------------------------------------
  ERROR: Failed building wheel for pyk4a
Failed to build pyk4a
ERROR: Could not build wheels for pyk4a which use PEP 517 and cannot be installed directly

lpasselin, 2020-09-27:
Hi, did you install the Kinect azure sdk?

If yes, read the instructions in the readme and make sure the paths on your
system correspond to the paths you provide to the pip command.

I hope this solves your problem.

ZhangLanTao, 2020-09-27:
> Hi, did you install the Kinect azure sdk? If yes, read the instructions in the readme and make sure the paths on your system correspond to the paths you provide to the pip command. I hope this solves your problem.
> […](#)

I had added “Azure Kinect SDK v1.4.0\sdk\include” and "Azure Kinect SDK v1.4.0\sdk\windows-desktop\amd64\release\bin" to both system PATH and user PATH. I followed every step of the instruction, and checked related paths several times. But the problem remains so I'm confused.

shagren, 2020-09-27:
Can you also add next option to pip: --no-use-pep517

-- Отправлено из мобильного приложения Яндекс.Почты

lpasselin, 2020-09-27:
Make sure you installed 1.4.0 and not 1.4.1. otherwise update the path.

Make sure the files are actually at the path by browsing there in the file
explorer.

ZhangLanTao, 2020-09-28:
@lpasselin The path do is right, I can find "k4a/k4a.h" in “C:\Program Files\Azure Kinect SDK v1.4.0\sdk\include“， 
[image]

ZhangLanTao, 2020-09-28:
@shagren It works, thank u very much~ BTW, why does this happen?

shagren, 2020-09-28:
@ZhangLanTao , there some links: https://github.com/etiennedub/pyk4a/pull/68#discussion_r492673320

In some step pip(or setuptools) lost your options `--global-option`

ZhangLanTao, 2020-09-29:
> @ZhangLanTao , there some links: [#68 (comment)](https://github.com/etiennedub/pyk4a/pull/68#discussion_r492673320)
> 
> In some step pip(or setuptools) lost your options `--global-option`

thank u veryvery much

closed 2020-09-29 by ZhangLanTao