etiennedub/pyk4a #78: Failed to get transformed depth frame on Ubuntu 18.04

Diananini, 2020-11-12:
I use playback to read mkv and want to get transformed depth frame. The code is as follows:
[image]

**It works well on Win10, but when run it on Ubuntu 18.04, I got an error**:
[2020-11-12 01:28:57.332] [error] [t=1300] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (1201): transformation_depth_image_to_color_camera_custom(transformation_handle, depth_image_buffer, &depth_image_descriptor, custom_image_buffer, &dummy_descriptor, transformed_depth_image_buffer, &transformed_depth_image_descriptor, transformed_custom_image_buffer, &dummy_descriptor, interpolation_type, invalid_custom_value) returned failure in k4a_transformation_depth_image_to_color_camera()
[2020-11-12 01:28:57.339] [error] [t=1300] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/transformation.c (584): k4a_transformation_t_get_context(). Invalid k4a_transformation_t (nil)
[2020-11-12 01:28:57.339] [error] [t=1300] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/transformation.c (649): Invalid argument to transformation_destroy(). transformation_handle ((nil)) is not a valid handle of type k4a_transformation_t

The version of pyk4a and Azure Kinect SDK on Ubuntu is the same as Win10.
How can I solve this?

lpasselin, 2020-11-12:
Can you confirm the pyk4a version is the same? With pip freeze

Also was the playback saved on windows and you are trying to reuse it on Linux? I am not sure if we tested this use case. 

I do not have a lot of experience with the playback. But can't you do capture.transformed_depth directly?

Diananini, 2020-11-12:
I checked the pyk4a version on win10 and ubuntu and both are v1.0.1. 
I recorded the video stream on Win10 with the tool k4aviewer.exe provided by Azure Kinect SDK, and I want to process the video on my ubuntu server. As you said, maybe that's the possible reason.
I also tried capture.transformed_depth, but I got error on both Win10 and ubuntu:

Traceback (most recent call last):
  File "show_depth_to_color.py", line 53, in <module>
    if capture.transformed_depth is not None:
  File "D:\Users\Diana\anaconda3\envs\kinect\lib\site-packages\pyk4a\capture.py", line 50, in transformed_depth
    self._transformed_depth = depth_image_to_color_camera(self._depth, self._calibration, self.thread_safe)
  File "D:\Users\Diana\anaconda3\envs\kinect\lib\site-packages\pyk4a\transformation.py", line 16, in depth_image_to_color_camera
    calibration.transformation_handle, thread_safe, depth, calibration.color_resolution,
AttributeError: 'NoneType' object has no attribute 'transformation_handle'

lpasselin, 2020-11-13:
Are you able to reproduce with a short playback that you could send us?

@shagren what do you think

Diananini, 2020-11-13:
I uploaded a short playback to Google drive:
https://drive.google.com/file/d/1iViprS0MdHJR2a8K5a0dUARJdhiRjps6/view?usp=sharing
Thanks for your help!
PS: I use Azure Kinect SDK v1.4.1 and pyk4a v1.0.1

shagren, 2020-11-17:
I cannot reproduce on my laptop but i can catch similar errors in CI: https://github.com/etiennedub/pyk4a/pull/82/checks?check_run_id=1413702228

shagren, 2020-11-17:
Next code works for me without errors with your sample:
```python
def play(playback: PyK4APlayback):
    while True:
        try:
            capture = playback.get_next_capture()
            if capture.color is not None:
                cv2.imshow("Color", convert_to_bgra_if_required(playback.configuration["color_format"], capture.color))
            if capture.depth is not None:
                cv2.imshow("Depth", colorize(capture.depth, (None, 5000)))
                transformed_depth = depth_image_to_color_camera(capture.depth, playback.calibration, False)
                cv2.imshow("TransformedDepth", colorize(transformed_depth, (None, 5000)))
            key = cv2.waitKey(10)
            if key != -1:
                break
        except EOFError:
            break
    cv2.destroyAllWindows()
```

Debian Stable, Python 3.7.3, pyk4a 1.0.1

shagren, 2020-11-17:
@Diananini , did you run code on Ubuntu  with desktop environment?

Diananini, 2020-11-18:
No. I run code on my Ubuntu server via SSH and use Xmanager to visualize the results.

shagren, 2020-11-18:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/810
Like as there  some glue. You need opengl (via X server?) for depth functionality.
It's Azure Kinekt SDK feature, not pyk4a.

shagren, 2020-11-18:
@lpasselin ,
how to reproduce: 
```
env DISPLAY= make test-no-hardware
```

and wou will see:
```
tests/functional/test_calibration.py::TestCalibration::test_creating_transfromation_handle PASSED                                                                                                                                [  5%][2020-11-18 12:07:03.014] [error] [t=97593] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/transformation.c (584): k4a_transformation_t_get_context(). Invalid k4a_transformation_t (nil)
[2020-11-18 12:07:03.014] [error] [t=97593] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/transformation.c (649): Invalid argument to transformation_destroy(). transformation_handle ((nil)) is not a valid handle of type k4a_transformation_t

```

closed 2020-11-19 by lpasselin