etiennedub/pyk4a #80: DLL load failed while importing k4a_module

ozgucbertug, 2020-11-14:
Building the library works fine however when I try to run the example code on Windows, I receive ImportError: DLL load failed while importing k4a_module: The specified module could not be found. Seems like #60 issues still persist. I am using python version 3.8.5

lpasselin, 2020-11-14:
Did you try with pip option --no-use-pep517 ?

Also what is your pip version?

ozgucbertug, 2020-11-15:
I've tried with the option --no-use-pep517 on pip version 20.2.4. Today I've created a new environment and re-entered the DLL directory to PATH. This solved the issue.

closed 2020-11-15 by ozgucbertug

bibeglimbu, 2021-04-08:
Hello,
I am having this same issue how ever no matter how many times i recreate the virtual env, it doesnt work....any details?