etiennedub/pyk4a #84: Misalignment between depth and RGB after transforming depth to RGB camera

jsaisagar, 2020-12-02:
Hi,

I am getting misalignment between RGB and depth outputs after "using capture.transformed_depth". Any inputs on how to resolve this issue?

Thanks,
Sai

lpasselin, 2020-12-02:
Would you mind sharing example images for RGB, depth and transformed_depth?

jsaisagar, 2020-12-03:
Hi Lpasselin,

I have used low light just to capture misalignment on edges. However, same is observed with sufficient lighting. I am attaching RGB and transformed depth.

[image]
[image]

Thanks for any help.

lpasselin, 2020-12-03:
I think this is normal behavior unfortunately. The transformed_depth seems to be doing it's job.

The calibration is not perfect and maybe you can refine it with a custom calibration.

@etiennedub can you confirm?

jsaisagar, 2020-12-03:
May I know which is the best way to refine calibration between RGB and Depth cameras to reduce this misalignment?

Thanks,

closed 2020-12-07 by lpasselin