etiennedub/pyk4a #85: How do I get a depth value from a depth image.

zhaocheng-hub, 2020-12-07:
Hello, 

how do I get the depth value at a certain point from the depth image that I get in my code?

How do I use this library to generate point cloud images?

Thanks,

lpasselin, 2020-12-07:
Hi, please see the various examples. You can browse the code of the library itself.
We use typical numpy arrays as image structure, which is very common with python.
Maybe you should look into opencv tutorials.

closed 2020-12-07 by lpasselin

zhaocheng-hub, 2020-12-08:
> 
> 
> Hi, please see the various examples. You can browse the code of the library itself.
> We use typical numpy arrays as image structure, which is very common with python.
> Maybe you should look into opencv tutorials.

Thank you very much for your advice.