etiennedub/pyk4a #86: Delay in captured images

joshpunb, 2020-12-08:
When additional analysis is applied to the image frame following the capture in the viewer.py example such that the looping frequency is less than 30 Hz, a noticeable delay gets introduced (~0.5 seconds) in the video. This makes it difficult to properly align the Kinect data with other external sensors.  

Perhaps this relates to the "lazily fetched" images. Is it possible to get the most recent capture using get_capture() and avoid generating a delay when the looping frequency is less than the camera fps?

lpasselin, 2020-12-08:
Unfortunately right now you must fetch the captures. This is also how the Azure Kinect SDK works iirc.
You do not need to fetch the images in the capture. Only the capture itself.

If you want to skip captures, you can set a timeout to the get_capture function.

You should probably use a worker object, like the CameraWorker in this example: https://github.com/etiennedub/pyk4a/blob/master/example/threads.py
Your CameraWorker could be constantly waiting for new captures and dropping older ones without processing them or fetching the frames.

Some tips for speed: make sure you understand the synchronized frames argument (see SDK docs). Also, for your purpose maybe the thread_safe option should be False for a slightly faster pyk4a.

joshpunb, 2020-12-09:
Thanks! The CameraWorker solved my issue.

closed 2020-12-14 by lpasselin