etiennedub/pyk4a #95: get specific frame from mkv by playback.py

JosephPai, 2021-01-13:
Hi,

Thanks for your awesome work!
I wonder if there are any way to get one specific frame from mkv by _**playback.py**_ ?
For now, I can only get the target frame by using _**get_next_capture()**_ tons of times.
I noticed that there is a method **_seek()_** in _**playback.py**_, but it returns nothing.
Also, it dose not change the pointer position of playback from my observation.

So, what is exactly the function the **_seek()_**?
And, if there are any way to get the target frame quickly?

Thanks!

shagren, 2021-01-13:
@JosephPai , you can seek by time in usec(microseconds), i.e. 

```python
  playback.seek(2000000) # seek to 2 seconds from start
  frame = playback.get_next_capture() 
```

You cannot seek to frame by number, but you can do some math tricks:

```
  usec_per_frame = 1 / 30 * 1E6 #  30 fps
  offset = (frame_number - 1) * usec_per_frame
  playback.seek(int(offset))
```

It won't be accurate enough, but I do not know better and faster way.

JosephPai, 2021-01-13:
That's helpful!
Thanks for your reply!
I'll close this issue.

closed 2021-01-13 by JosephPai