etiennedub/pyk4a #97: Exposure auto mode affecting other parameters

chrismeunier, 2021-01-19:
I found some strange behaviors while experimenting with the auto mode for exposure and whitebalance:
1. If I start a camera without setting exposure and whitebalance (full auto) when querying their values I always get 33330μs and 4500K.
2. In auto exposure mode, manually changing the gain value doesn't have any effect. 
3. When changing from auto exposure to a manual value of 33330μs (max for 30fps) and then manually setting the gain, the overall intensity of the image decreases significantly (even with max gain). This is easily visible in a dark room with ambient lighting. In a bright room during the day this is not necessarily visible.

To understand 3. I tried manually setting the gain value to its maximum (255) to balance this effect. It kind of works but setting the exposure to auto again will still give a brighter image than a manual setting of max exposure and max gain.

To reproduce the problems:
1. Start in auto exposure and whitebalance. Or change the values, then go back to auto, the values you get from the camera are the last ones set manually.
2. Start or set the exposure to auto mode. Try changing the gain value.
3. Set the parameters to match approximately what you get in auto mode (typically max gain and exposure and some appropriate whitebalance). Go back to auto whitebalance and exposure, changing the gain now has no effect. Set the exposure manually, if you keep it at the max value the image should still look the same. Change the gain value, the overall brightness should now change.

So for some reason the auto exposure seems to affect the gain and some other hidden parameter. The brightness doesn't seem to be it, trying to change it doesn't affect the image the same way the auto mode does. If you have any idea or question let me know, all I found was [this vague mention of gain.](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1281)

Here I include a slightly modified version of your viewer.py file which uses some keys to modify the parameters and displays their values on the image for easy testing:
```python
import cv2
import numpy as np

import pyk4a
from pyk4a import PyK4A

""" Minimal working example based on example/viewer.py
    Whitebalance, exposure time, gain and brightness can be changed or set
    to auto when running using keys. Esc or 'q' exit the program.

    (+) and (-) keys for settings:
    '.' and ',' for whitebalance, 
    'm' and 'n' for exposure,
    'g' and 'f' for gain,
    'b' and 'v' for brightness.
"""

def main():
    k4a = PyK4A() # all default parameters: 720p, 30fps...
    k4a.start()

    # getters and setters directly get and set on device 
    k4a.whitebalance = 4500
    # dict of exposure values (for 30fps):
    exp_dict = {-11: 500, -10: 1250, -9: 2500, -8: 8330, -7: 16670, -6: 33330}
    exp_val = -6 # to be changed when running
    k4a.exposure = exp_dict[exp_val]
    # (whitebalance & exposure are manual now)

    # text settings
    text_font = cv2.FONT_HERSHEY_SIMPLEX
    text_scale = 1
    text_clr = (255, 255, 255)

    while 1:
        capture = k4a.get_capture()
        if np.any(capture.color):
            img = capture.color[:, :, :3].copy() # cannot write text on capture
            # Add "debug" text on display
            text_line1 = "Whitebalance:{} (auto: {})".format(
                k4a.whitebalance, k4a.whitebalance_mode_auto)
            cv2.putText(img, text_line1, (10,50), 
                        text_font, text_scale, text_clr)
            text_line2 = "Exposure:{} (auto: {})".format(
                k4a.exposure, k4a.exposure_mode_auto)
            cv2.putText(img, text_line2, (10, 100),
                        text_font, text_scale, text_clr)
            text_line3 = "Gain:{}, Brightness:{}".format(
                k4a.gain, k4a.brightness)
            cv2.putText(img, text_line3, (10, 150),
                        text_font, text_scale, text_clr)

            cv2.imshow("k4a", img)
            key = cv2.waitKey(10)

            if key == 27 or key == 113: # 'Esc' or 'q' to close
                cv2.destroyAllWindows()
                break
            # New key commands:
            elif key == 97:  # 'a' resets to auto mode
                k4a.whitebalance_mode_auto = True
                k4a.exposure_mode_auto = True
            # Whitebalance
            elif key == 46:  # increase with "."
                k4a.whitebalance = min(12500, k4a.whitebalance + 100)
            elif key == 44:  # decrease with ","
                k4a.whitebalance = max(2500, k4a.whitebalance - 100)
            # Exposure            
            elif key == 109:  # "m" to increase
                exp_val = min(-6, exp_val + 1)
                k4a.exposure = exp_dict[exp_val]
            elif key == 110:  # "n" to decrease
                exp_val = max(-11, exp_val - 1)
                k4a.exposure = exp_dict[exp_val]
            # Gain
            elif key == 103:  # 'g' to increase
                k4a.gain = min(255, k4a.gain + 1)
            elif key == 102:  # 'f' to decrease
                k4a.gain = max(0, k4a.gain - 1)
            # Brightness (not really necessary)
            elif key == 98:  # 'b' to increase
                k4a.brightness = min(255, k4a.brightness + 1)
            elif key == 118: # 'v' to decrease
                k4a.brightness = max(0, k4a.brightness - 1)
    k4a.stop()

if __name__ == "__main__":
    main()
```

My setup: Windows 10, using SDK v1.4.1, Kinect firmware AzureKinectDK_Fw_1.6.110079014, Python 3.8.6, pyk4a 1.1.0.
And btw thanks a lot for your work, it really helps a ton!

lpasselin, 2021-01-19:
Interesting bug.

Is it possible that we wrongly assume the exposure returned during auto exposure mode is valid? Maybe it is not valid and we should not be able to read exposure in auto mode.

Also we have to check if this happens with the official sdk viewer.

lpasselin, 2021-03-02:
Did you have any new information for this bug?

chrismeunier, 2021-03-09:
Not really sadly, I tested it with the official kinect viewer and I can reproduce the same inconsistencies with the gain and exposure values in auto or manual mode.

I also found some issues on the SDK repository:
- An [open issue](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1378) (old) with the exact same bug described and no suggestion from the developers yet.
- An [even weirder bug](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1511) where starting with manual exposure values just doesn't work. I do not have this problem since I switch to manual values after starting the camera.
- More interesting: there seems to be [a way to get the exposure value](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1055#issuecomment-582055446) when set to auto. **Is this available with pyk4a?** I would be curious to know if this exists for white balance too...

Overall it looks like one bug (or more) to be solved on the SDK side.

lpasselin, 2021-03-31:
I will close as this is a problem with the SDK

closed 2021-03-31 by lpasselin