etiennedub/pyk4a #99: Camera calibration parameter

zhaocheng-hub, 2021-01-22:
Hello, 
I got the camera's internal and external parameters "mtx", "dist", "rvecs", "tvecs" of color and depth camera through OpenCV。 
How to write these parameters to the function calibration matrix of the project application, and how to use these values.

lpasselin, 2021-01-22:
This could be documented better. Both in our wrapper and on the official SDK.

See functions `save_calibration_json`and `load_calibration_json`
https://github.com/etiennedub/pyk4a/blob/6d1355a1ee994fddcabb48501586b058671c9443/pyk4a/pyk4a.py#L74
https://github.com/etiennedub/pyk4a/blob/6d1355a1ee994fddcabb48501586b058671c9443/pyk4a/pyk4a.py#L78

You should use `save_calibration_json` first to have a template.
An example can be found here: https://github.com/etiennedub/pyk4a/blob/6d1355a1ee994fddcabb48501586b058671c9443/tests/assets/calibration.json

closed 2021-01-22 by lpasselin

zhaocheng-hub, 2021-01-23:
> 可以更好地记录下来。 无论是在我们的包装中还是在官方SDK中。
> 
> 查看功能 `save_calibration_json`和 `load_calibration_json`
> 
> https://github.com/etiennedub/pyk4a/blob/6d1355a1ee994fddcabb48501586b058671c9443/pyk4a/pyk4a.py#L74
> 
> https://github.com/etiennedub/pyk4a/blob/6d1355a1ee994fddcabb48501586b058671c9443/pyk4a/pyk4a.py#L78
> 
> 您应该 使用 `save_calibration_json`首先 一个模板。
> 可以在这里找到一个示例： : [https //github.com/etiennedub/pyk4a/blob/6d1355a1ee994fddcabb48501586b058671c9443/tests/assets/calibration.json ](https://github.com/etiennedub/pyk4a/blob/6d1355a1ee994fddcabb48501586b058671c9443/tests/assets/calibration.json)

I try to change some data manually, and try to use  `k4a. load_ calibration_ json`  to read the data of "JSON" in the path I specified. And `k4a.calibration_raw`  is printed with `print`  There is no change in the attribute of ` k4a.calibration_raw`.

In addition, through  `k4a.save_calibration_json`  saves "JSON" data, it is effective. It changes the original "json" data (but the style becomes only one line of code instead of the style of the template you gave me).

I will `k4a.load_calibration_json` in the main function, I found that he successfully printed my modified "json", but this method still can't change `k4a.calibration_raw`

Maybe my usage is wrong. Can you give me an example?

shagren, 2021-01-23:
You can use high level api: create Calibration from string and then use it for image transformation or coordinates translation
```python
calibration = Calibration.from_raw(json_string, DepthMode.NFOV_UNBINNED, ColorResolution.RES_1080P)
transormed_image =  depth_image_to_color_camera(depth_image, calibration) 
color_x, color_y, color_z  = calibration.depth_to_color_3d((depth_x, depth_y, depth_z)) # 3d coordinates from Depth camera space to Color camera space
```

zhaocheng-hub, 2021-01-23:
> 您可以使用高级api：根据字符串创建校准，然后将其用于图像转换或坐标转换
> 
> ```python
> calibration = Calibration.from_raw(json_string, DepthMode.NFOV_UNBINNED, ColorResolution.RES_1080P)
> transormed_image =  depth_image_to_color_camera(depth_image, calibration) 
> color_x, color_y, color_z  = calibration.depth_to_color_3d((depth_x, depth_y, depth_z)) # 3d coordinates from Depth camera space to Color camera space
> ```

 Thank you for your reply， this method is effective.