etiennedub/pyk4a #100: Missing conversion from 3d to 2d and from 2d to 2d

leonardokidd, 2021-01-26:
Hi,
May I ask if the conversion from 3d to 2d and 2d to 2d are intentional left out or have not been implemented yet? Is there any plan of adding these useful functions?

Appreciate for the awesome work, and thanks in advance!

lpasselin, 2021-01-26:
Hi, you are right they have not been added yet. There is no plan but we would gladly accept a PR for this!

You can use the functions depth_to_color and color_to_depth in transformation.py (or follow Capture.transformed_depth)

leonardokidd, 2021-01-26:
Thanks for the quick response. The functions you mentioned are essentially for conversion between 2D depth and 2D RGB images if I understand correctly. I'll see what I can do for the PR as I'm not very familiar with Cython. Thanks for the explanation.

closed 2021-04-07 by lpasselin