etiennedub/pyk4a #103: Visualizing local images through  'color_image_to_depth_camera'

zhaocheng-hub, 2021-03-01:
Hello, I encountered a problem when using method "a" to get "Transformed Color"  image from local image.   
My code is as follows ， Depth image and color image are normal .    The image  'Transformed Color'  is abnormal. 

`python

    k4a.start()

    depth = cv2.imread('../picture/test/depth/300.png')
    color = cv2.imread('../picture/test/color/300.png')

    capture = k4a.get_capture()
 
    if depth is not None:  
        cv2.imshow("Depth_colorize", colorize(depth, (None, 5000)))
    if color is not None:  
        cv2.imshow("Color", color[:, :, :3])
    if pyk4a.color_image_to_depth_camera(color, depth, capture._calibration,
                                         capture.thread_safe) is not None:  
        cv2.imshow("Transformed Color",
                   pyk4a.color_image_to_depth_camera(color, depth, capture._calibration, capture.thread_safe)[:, :, :3])
    cv2.waitKey(10)

    k4a.stop()
`
[image]

In addition, I have used the method 'depth_image_to_point_cloud'  in the same form, and a similar problem has occurred. I think it should be the same reason.

lpasselin, 2021-03-02:
I am not sure what the problem is.

Can you try passing `color[:, :, :3]` to `color_image_to_depth_camera` ? Just like you did with imshow?

Similarly, make sure the depth image is in the same format as the captured depth. Same for the color image.

If that doesn't work, can you share the images here to help me debug?

zhaocheng-hub, 2021-03-02:
Hello, I confirm that the format and capture time of depth image and color image are the same. If they are different, it will produce the following feedback.

`[2021-03-02 09:49:22.958] [error] [t=6125] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/rgbz.c (113): transformation_compare_image_descriptors(). Unexpected image descriptor. Descriptor 1: width_pixels: 640, height_pixels: 576, stride_bytes: 2560, format: 3. Descriptor 2: width_pixels: 512, height_pixels: 512, stride_bytes: 2048, format: 3. `

`[2021-03-02 09:49:22.958] [error] [t=6125] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/rgbz.c (986): transformation_color_image_to_depth_camera_validate_parameters(). Unexpected transformed color image descriptor, see details above.`

`[2021-03-02 09:49:22.958] [error] [t=6125] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (1267): transformation_color_image_to_depth_camera(transformation_handle, depth_image_buffer, &depth_image_descriptor, color_image_buffer, &color_image_descriptor, transformed_color_image_buffer, &transformed_color_image_descriptor) returned failure in k4a_transformation_color_image_to_depth_camera()`

I'm using method 'depth_image_to_point_cloud' in the same way and it's producing the wrong result.
`points = pyk4a.depth_image_to_point_cloud(depth, capture._calibration, capture.thread_safe, calibration_type_depth=True,).reshape((-1,3))`  I think they should have similar problems.  

I made some attempts. When capturing data , I save the 'capture.depth_point_cloud' and ' capture.transformed_color ' data through the following code. Then visualize and get the correct image.

` np.save(path+'depth_point_cloud/'+str(num),capture.depth_point_cloud, allow_pickle=True, fix_imports=True)`
  `np.save(path+'transformed_color/'+str(num),capture.transformed_color, allow_pickle=True, fix_imports=True)`

The images I use are as follows：
[image]
[image]

I'm using the following pattern.

` k4a = PyK4A(
        Config(
            color_resolution=pyk4a.ColorResolution.RES_720P,
            camera_fps=pyk4a.FPS.FPS_15,
            depth_mode=pyk4a.DepthMode.NFOV_UNBINNED,  
            synchronized_images_only=True,  
        )
    )`

Thank you for your help.

lpasselin, 2021-03-02:
Maybe the internal c arrays of the imread images do not follow the same stride as the kinect azure images. Can you compare `color.strides` of both images?

I don't have much time this week but will help in this thread. Next week when I have access to my setup I'll test it.

zhaocheng-hub, 2021-03-03:
Next time, I will try to save `color` and `depth` images in `npy`  format, and try the method ' color_image_to_depth_camera'  .  If there are any new results, I will let you know. Thank you.

shagren, 2021-03-03:
@zhaocheng-hub , we use `k4arecorder` in our work.
`pyk4a` can read mkv files created by  k4arecorder, see [example](https://github.com/etiennedub/pyk4a/blob/master/example/playback.py). Captures  from mkv identical to captures from camera.

zhaocheng-hub, 2021-03-03:
@shagren  Thank you for your advice. 
Next time I'll try to capture images from local MKV and try method `color_image_to_depth_camera`, I'll give you the results of my attempt.

zhaocheng-hub, 2021-03-04:
@lpasselin  Hello, I try to save the image in `npy` format, and then get the `transformed_color` image through the method `color_image_to_depth_camera` .   The 'transformed_color' image is obtained successfully. I want to know why？

    depth = np.load('../picture/test/depth/501.npy')
    color = np.load('../picture/test/color/501.npy')

    capture = k4a.get_capture()

    if depth is not None:  
        cv2.imshow("Depth_colorize", colorize(depth, (None, 5000)))
    if color is not None:  
        cv2.imshow("Color", color[:, :, :3])
    if pyk4a.color_image_to_depth_camera(color, depth, capture._calibration,
                                         capture.thread_safe) is not None:  
        cv2.imshow("Transformed Color",
                   pyk4a.color_image_to_depth_camera(color[:,:,:3], depth, capture._calibration, capture.thread_safe)[:, :, :3])
    cv2.waitKey(1000)

[image]

zhaocheng-hub, 2021-03-04:
@shagren  Hello，
I try to use 'playback' to read the `mkv` file and use 'color_image_to_depth_camera' to get the `transformed_color` image.  `color` Image  and image 'depth' can be visualized ，  But I encountered an error. My key code is as follows.   

`[2021-03-04 23:12:05.076] [error] [t=13110] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (66): Invalid argument to image_create_from_buffer(). height_pixels <= 0 || height_pixels > 20000`
`[2021-03-04 23:12:05.076] [error] [t=13110] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (66): image_create_from_buffer() returned failure.`

def play(playback: PyK4APlayback):

        while True:
            try:
                capture = playback.get_next_capture()
                if capture.color is not None:
                    cv2.namedWindow("Color", 0)
                    cv2.resizeWindow("Color", 1280, 960)
                    cv2.moveWindow("Color", 0, 0)
                    cv2.imshow("Color",
                               convert_to_bgra_if_required(playback.configuration["color_format"], capture.color))
                if capture.depth is not None:
                    cv2.namedWindow("Depth", 0)
                    cv2.resizeWindow("Depth", 960, 720)
                    cv2.moveWindow("Depth", 1000, 0)
                    cv2.imshow("Depth", colorize(capture.depth, (None, 5000)))
                if pyk4a.color_image_to_depth_camera(capture.color, capture.depth, capture._calibration,
                                                     capture.thread_safe) is not None:  # 彩色转深度格式的 加上深度点
                    cv2.imshow("Transformed Color",
                               pyk4a.color_image_to_depth_camera(capture.color, capture.depth, capture._calibration,
                                                                 capture.thread_safe)[:, :, :3])
                key = cv2.waitKey(10000)
                if key != -1:
                    break
            except EOFError:
                break
        cv2.destroyAllWindows()

shagren, 2021-03-04:
Maybe your color image not in rgba format? Please check playback.configuration["color_format"]

In this case you need convert color first(solution for mjpeg):
```
color_image = cv2.imdecode(capture.color, cv2.IMREAD_UNCHANGED) 
rgba_color_image = cv2.cvtColor(color_image, cv2.COLOR_RGB2RGBA)
```
Then you can use `rgba_color_image` in `color_image_to_depth_camera()`

zhaocheng-hub, 2021-03-05:
@shagren  Thank you.

 He's very efficient and the method `color_image_to_depth_camera'`is very successful.

I still want to know that using method `color_image_to_depth_camera` by directly reading the local image is a failure. The array of `NPY` format and the method you mentioned are feasible.

lpasselin, 2021-03-09:
@zhaocheng-hub Compare the `strides` and `dtype` of the images before and after saving with opencv.

closed 2021-03-09 by lpasselin

zhaocheng-hub, 2021-03-10:
@lpasselin  Thank you for your.