etiennedub/pyk4a #108: Get data values from depth image

NNNTAn, 2021-03-07:
Hello,
May I ask about the coordinates of depth image?
I would like to get the depth coordinates to measure the object. I know the data of depth image is in mm, but how can I get it?
Thank you,

lpasselin, 2021-03-07:
Hi, I'm not sure what you don't understand. To get the depth images you get a capture and then call capture.depth.
The depth data is contained in the image. Each pixel corresponds to a depth measurement, in mm.
See the examples folder scripts.

NNNTAn, 2021-03-07:
> Hi, I'm not sure what you don't understand. To get the depth images you get a capture and then call capture.depth.
> The depth data is contained in the image. Each pixel corresponds to a depth measurement, in mm.
> See the examples folder scripts.

Thank you for your response!
I already know how to capture color and depth images. 
After that I transformed the depth image to the color image to find the coordinate of point of interest from color image.
I would like to get the corresponding depth values of those points from the transformed depth image. Could you help me how to call the function to do that or a sample code.
Thank you,

closed 2021-03-08 by NNNTAn