etiennedub/pyk4a #111: How do I reference the 'pyk4a' library

zhaocheng-hub, 2021-03-10:
Hello, I have used some methods of 'pyk4a' library in my academic paper. How should I quote them in my paper?

lpasselin, 2021-03-10:
You could use this:
````
@misc{pyk4a,
  author = {Louis-Philippe Asselin and Etienne Dubeau and Ilya Gruzinov},
  title = {pyk4a},
  year = {2020},
  publisher = {GitHub},
  journal = {GitHub repository},
  howpublished = {\url{https://github.com/etiennedub/pyk4a/}},
}
````

Or cite our papers (they do not mention the library but we developed pyk4a for them initially):
https://lvsn.github.io/real-svbrdf/
https://lvsn.github.io/rgbde_tracking/

Or footnote with url of the repository.

My Master's thesis briefly introduces pyk4a but it will only be published in a couple of weeks.

lpasselin, 2021-03-10:
@shagren can you confirm you are ok with having your name in the citation?

closed 2021-03-10 by lpasselin

shagren, 2021-03-11:
@lpasselin , confirmed

zhaocheng-hub, 2021-03-12:
@lpasselin  ok