etiennedub/pyk4a #117: Skipping wheel build for pyk4a, due to binaries being disabled for it.

ZhengyangSun1020, 2021-04-09:
Hi, I used the pip install pyk4a --no-use-pep517 --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib" 
And get this error:
D:\download\avaconda\envs\py39\lib\site-packages\pip\_internal\commands\install.py:230: UserWarning: Disabling all use of wheels due to the use of --build-option / --global-option / --install-option.
  cmdoptions.check_install_build_global(options)
Collecting pyk4a
  Using cached pyk4a-1.2.3.tar.gz (20 kB)
Collecting numpy
  Using cached numpy-1.20.2.zip (7.8 MB)
Requirement already satisfied: python_version>="3.4" in d:\download\avaconda\envs\py39\lib\site-packages (from pyk4a) (0.0.2)
Skipping wheel build for pyk4a, due to binaries being disabled for it.
Skipping wheel build for numpy, due to binaries being disabled for it.
Installing collected packages: numpy, pyk4a
    Running setup.py install for numpy ... error

ZhengyangSun1020, 2021-04-09:
And I meet the issue like
when importing:
from pyk4a import PyK4A
I hit the issue:
"DLL load failed while importing k4a_module: The specified module could not be found"

ZhengyangSun1020, 2021-04-09:
the last environment I used is python 3.9
When I changed to python3.8, use  pip install pyk4a --no-use-pep517 --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"
**Get the same issue:**
D:\download\avaconda\lib\site-packages\pip\_internal\commands\install.py:230: UserWarning: Disabling all use of wheels due to the use of --build-option / --global-option / --install-option.

lpasselin, 2021-04-09:
There were changes yesterday in the pypi package, thanks a lot for reporting. I am checking this right now.

lpasselin, 2021-04-09:
I read your information too quickly.
The issue is not related to the PyPi package. It is related to your Windows environment not finding the DLL.
To be certain, you could you try installing `pyk4a==1.1.0` but I doubt that will make any difference.

`UserWarning: Disabling all use of wheels due to the use of --build-option / --global-option / --install-option.`
Is fine. You are not using wheels. We use sdist so you compile the C part of pyk4a yourself when installing pyk4a through pip. The compilation part is not working for you.

This seems to be the problem:
````
Installing collected packages: **numpy**, pyk4a`
Running setup.py install for numpy ... error
````

Can you try adding `--no-build-isolation` and/or removing `--no-use-pep517`?

lpasselin, 2021-04-09:
Also, make sure you have numpy installed before installing pyk4a.

ZhengyangSun1020, 2021-04-09:
Okay, I uninstall the pyk4a and install numpy. And then I tried to run like this:
pip install pyk4a --no-build-isolation --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"

The result is 
D:\download\avaconda\lib\site-packages\pip\_internal\commands\install.py:230: UserWarning: Disabling all use of wheels due to the use of --build-option / --global-option / --install-option.
  cmdoptions.check_install_build_global(options)
Collecting pyk4a
  Using cached pyk4a-1.2.3.tar.gz (20 kB)
    Preparing wheel metadata ... done
Requirement already satisfied: python-version>="3.4" in d:\download\avaconda\lib\site-packages (from pyk4a) (0.0.2)
Requirement already satisfied: numpy in d:\download\avaconda\lib\site-packages (from pyk4a) (1.19.2)
Building wheels for collected packages: pyk4a
  Building wheel for pyk4a (PEP 517) ... error

lpasselin, 2021-04-09:
Are you sure the Paths that you provide with the command are right?
`--global-option="-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"`

If there is an error these, I would suggest retrying from the start, with the command in the README (with your custom paths)

Also, do you see any other error logs that could help us find the problem?

ZhengyangSun1020, 2021-04-09:
Yep, I'm sure the path is right.....
I will try again

shagren, 2021-04-09:
worked for me with  --no-use-pep517 :
```
>pip install pyk4a  --no-use-pep517  --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"
c:\users\azure\pycharmprojects\xxx\venv\lib\site-packages\pip\_internal\commands\install.py:230: UserWarning: Disabling all use of wheels due to the use of --build-option / --global-option / --install-option.
  cmdoptions.check_install_build_global(options)
Collecting pyk4a
  Using cached pyk4a-1.2.3.tar.gz (20 kB)
Requirement already satisfied: numpy in c:\users\azure\pycharmprojects\xxx\venv\lib\site-packages (from pyk4a) (1.20.2)
Requirement already satisfied: python_version>="3.4" in c:\users\azure\pycharmprojects\xxx\venv\lib\site-packages (from pyk4a) (0.0.2)
Skipping wheel build for pyk4a, due to binaries being disabled for it.
Installing collected packages: pyk4a
    Running setup.py install for pyk4a ... done
Successfully installed pyk4a-1.2.3
```

ZhengyangSun1020, 2021-04-09:
Oh...When I use the code in readme again. It works good.
D:\download\avaconda\lib\site-packages\pip\_internal\commands\install.py:230: **UserWarning: Disabling all use of wheels due to the use of --build-option / --global-option / --install-option.**
  cmdoptions.check_install_build_global(options)
Collecting pyk4a
  Using cached pyk4a-1.2.3.tar.gz (20 kB)
Requirement already satisfied: numpy in d:\download\avaconda\lib\site-packages (from pyk4a) (1.19.2)
Requirement already satisfied: python_version>="3.4" in d:\download\avaconda\lib\site-packages (from pyk4a) (0.0.2)
Skipping wheel build for pyk4a, due to binaries being disabled for it.
Installing collected packages: pyk4a
    Running setup.py install for pyk4a ... done
Successfully installed pyk4a-1.2.3

But sorry to trouble again, when I run   from pyk4a import PyK4A
the result is : ImportError: DLL load failed while importing k4a_module: The specified module could not be found.

shagren, 2021-04-09:
@ZhengyangSun1020 ,

I have next code for windows in some module
```
def _add_dll_directory(path: Path):
    from ctypes import c_wchar_p, windll  # type: ignore
    from ctypes.wintypes import DWORD

    AddDllDirectory = windll.kernel32.AddDllDirectory
    AddDllDirectory.restype = DWORD
    AddDllDirectory.argtypes = [c_wchar_p]
    AddDllDirectory(str(path))

def kinect():
    if sys.platform != "win32":
        return
    env_path = os.getenv("KINECT_LIBS", None)
    if env_path:
        candidate = Path(env_path)
        dll = candidate / "k4a.dll"
        if dll.exists():
            _add_dll_directory(candidate)
            return
    # autodetecting
    program_files = Path("C:\\Program Files\\")
    for dir in sorted(program_files.glob("Azure Kinect SDK v*"), reverse=True):
        candidate = dir / "sdk" / "windows-desktop" / "amd64" / "release" / "bin"
        dll = candidate / "k4a.dll"
        if dll.exists():
            _add_dll_directory(candidate)
            return
```

You need to call  `kinect()` before importing from `pyk4a`

@lpasselin , maybe it's time to add this trick to package or readme because all love funnyng windows?

lpasselin, 2021-04-09:
Check out the thread and this comment https://github.com/etiennedub/pyk4a/pull/68#issuecomment-701526325

lpasselin, 2021-04-09:
@shagren I agree but I am unable to test any of this. I don't have a windows machine.

We should at least add a `try except` with a link to a FAQ that explains the situation?

I can't believe all python programs are forced to do this on windows!
Maybe we are doing something wrong during install

ZhengyangSun1020, 2021-04-09:
Sorry, I'm the beginner with too many problems 
when I add this code before:
`def _add_dll_directory(path: Path):
    from ctypes import c_wchar_p, windll  # type: ignore
    from ctypes.wintypes import DWORD

    AddDllDirectory = windll.kernel32.AddDllDirectory
    AddDllDirectory.restype = DWORD
    AddDllDirectory.argtypes = [c_wchar_p]
    AddDllDirectory(str(path))

def kinect():
    if sys.platform != "win32":
        return
    env_path = os.getenv("KINECT_LIBS", None)
    if env_path:
        candidate = Path(env_path)
        dll = candidate / "k4a.dll"
        if dll.exists():
            _add_dll_directory(candidate)
            return
    # autodetecting
    program_files = Path("C:\\Program Files\\")
    for dir in sorted(program_files.glob("Azure Kinect SDK v*"), reverse=True):
        candidate = dir / "sdk" / "windows-desktop" / "amd64" / "release" / "bin"
        dll = candidate / "k4a.dll"
        if dll.exists():
            _add_dll_directory(candidate)
            return`

But it says the `path` is not defined

ZhengyangSun1020, 2021-04-09:
> @shagren I agree but I am unable to test any of this. I don't have a windows machine.
> 
> We should at least add a `try except` with a link to a FAQ that explains the situation?
> 
> I can't believe all python programs are forced to do this on windows!
> Maybe we are doing something wrong during install

Which os do you use? Mac or Linux?

ZhengyangSun1020, 2021-04-09:
Meantime, Do I need to define `os` and `sys`? and how?

lpasselin, 2021-04-09:
I have access to Mac and linux.

With shagren's code, I think you are missing `from pathlib import Path`

Other option:
If you're on python version >= 3.8 you can also try
````
import os
os.add_dll_directory("path of directory containing k4a.dll")
import pyk4a
````

ZhengyangSun1020, 2021-04-09:
Thank you! I make it when adding the code from @shagren 

Thank you for helping me with patience!

I'm the beginner and try to use azure kinect. I guess I will trouble you again in the future about it! Thank you again!

closed 2021-04-09 by ZhengyangSun1020