etiennedub/pyk4a #118: cannot import name 'connected_device_count' from 'pyk4a'

bibeglimbu, 2021-04-09:
Hello, I ran into this problem when i tried to get the number of devices count. I copied it from your example but it didnt work. I used the below snipped to bypass the dll loading (k4a_module) problem.

from ctypes import c_wchar_p, windll  
from ctypes.wintypes import DWORD

AddDllDirectory = windll.kernel32.AddDllDirectory
AddDllDirectory.restype = DWORD
AddDllDirectory.argtypes = [c_wchar_p]
AddDllDirectory(r"C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\bin") # modify path there if required

import cv2
import numpy as np
import pyk4a
from pyk4a import Config, PyK4A

and i this is the part where it loads.

def  getDeviceCount():
    from pyk4a import connected_device_count
    cnt = connected_device_count()
    print(cnt)
Help greatly appreciated (beginner here)

lpasselin, 2021-04-09:
Can you show the error log?

bibeglimbu, 2021-04-09:
``(.venv)` PS D:\Programming\Git\Kinect\HelloKinect\Python> & d:/Programming/Git/Kinect/HelloKinect/Python/.venv/Scripts/python.exe d:/Programming/Git/Kinect/HelloKinect/Python/hello.py
Traceback (most recent call last):
  File "d:\Programming\Git\Kinect\HelloKinect\Python\hello.py", line 64, in <module>
    main()
  File "d:\Programming\Git\Kinect\HelloKinect\Python\hello.py", line 20, in main
    getDeviceCount()
  File "d:\Programming\Git\Kinect\HelloKinect\Python\hello.py", line 16, in getDeviceCount
    from pyk4a import connected_device_count
ImportError: cannot import name 'connected_device_count' from 'pyk4a' (D:\Programming\Git\Kinect\HelloKinect\Python\.venv\lib\site-packages\pyk4a\__init__.py)`

bibeglimbu, 2021-04-09:
sorry accidental closing :D , I hope that is what you mean. or are you looking for something else?

lpasselin, 2021-04-09:
Which version of pyk4a do you have? `connected_device_count` is only available since version `1.2.0`

bibeglimbu, 2021-04-09:
Name: pyk4a
Version: 1.1.0

I tried to update it but then it tan into this issue

pip install pyk4a --upgrade
Requirement already satisfied: pyk4a in d:\programming\git\kinect\hellokinect\python\.venv\lib\site-packages (1.1.0)
Collecting pyk4a
  Downloading pyk4a-1.2.3.tar.gz (20 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting python-version>="3.4"
  Downloading python_version-0.0.2-py2.py3-none-any.whl (3.4 kB)
Requirement already satisfied: numpy in d:\programming\git\kinect\hellokinect\python\.venv\lib\site-packages (from pyk4a) (1.20.2)
Building wheels for collected packages: pyk4a
  Building wheel for pyk4a (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: 'd:\programming\git\kinect\hellokinect\python\.venv\scripts\python.exe' 'd:\programming\git\kinect\hellokinect\python\.venv\lib\site-packages\pip\_vendor\pep517\_in_process.py' build_wheel 'C:\Users\bib_l\AppData\Local\Temp\tmp8n926rl8'
       cwd: C:\Users\bib_l\AppData\Local\Temp\pip-install-xza52pzl\pyk4a_1b0f27daaacb4079a93eb9f458091175 ....(there is more)

lpasselin, 2021-04-09:
Because you are on windows, there a bit more to add to the pip install command. See the readme of the repo for more info.

closed 2021-04-09 by bibeglimbu