etiennedub/pyk4a #119: Attempted relative import with no known parent package

ZhengyangSun1020, 2021-04-09:
Sorry, I have another problem that when I run this code 
`from pyk4a import Config, ImageFormat, PyK4ACapture, PyK4APlayback, PyK4ARecord`
the error is 
`attempted relative import with no known parent package`

ZhengyangSun1020, 2021-04-09:
There are many similar code with this error log like:
`from errors import K4AException, _verify_error`

it shows:
`ImportError: attempted relative import with no known parent package`

ZhengyangSun1020, 2021-04-09:
And btw, if I wanna track human body movement, can this realize?

ZhengyangSun1020, 2021-04-10:
Sorry to trouble you again

lpasselin, 2021-04-11:
The body tracking is not fully supported yet. I won't be able to Help you much with this.

But the problem is the installation

ZhengyangSun1020, 2021-04-11:
Okay, thank you!

closed 2021-04-11 by ZhengyangSun1020