etiennedub/pyk4a #120: How to diagnose K4AException for device.open() error

michhar, 2021-04-10:
Hello!  This is a great project and I want to thank you for making this.  I am getting an error in the last mile.

Summary:  I'm trying to test whether I can open my Kinect DK from an Ubuntu 18.04 docker container running on an ARM64 device with `pyk4a`.

Repro:
- Create an Ubuntu 18.04 docker container and run on ARM64 device (also running linux, a version of RHEL).
- Install Azure Kinect Sensor SDK according to the instructions on the GitHub repo [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#debian-package).  Version 1.4 and installed from inside the running docker container.
- Install `py4a` with `pip` with Python 3.6.9 (installed with `apt-get install python3-dev`).
- Run the example [`devices.py`](https://github.com/etiennedub/pyk4a/blob/master/example/devices.py) for device check.

Output and traceback:
```
Available devices: 1
---------------------------------------------------------------------------
K4AException                              Traceback (most recent call last)
<ipython-input-7-65eaaf62310e> in <module>
      6 for device_id in range(cnt):
      7     device = PyK4A(device_id=device_id)
----> 8     device.open()
      9     print(f"{device_id}: {device.serial}")
     10     device.close()

/usr/local/lib/python3.6/dist-packages/pyk4a/pyk4a.py in open(self)
     66         if self.opened:
     67             raise K4AException("Device already opened")
---> 68         self._device_open()
     69 
     70     def close(self):

/usr/local/lib/python3.6/dist-packages/pyk4a/pyk4a.py in _device_open(self)
     83     def _device_open(self):
     84         res, handle = k4a_module.device_open(self._device_id, self.thread_safe)
---> 85         _verify_error(res)
     86         self._device_handle = handle
     87 

/usr/local/lib/python3.6/dist-packages/pyk4a/errors.py in _verify_error(res)
     16     res = Result(res)
     17     if res == Result.Failed:
---> 18         raise K4AException()
     19     elif res == Result.Timeout:
     20         raise K4ATimeoutException()

K4AException: 
```

I also tried from the command line and Python interpreter inside the docker container:
```
Python 3.6.9 (default, Jan 26 2021, 15:33:00)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyk4a import PyK4A, connected_device_count
>>> cnt = connected_device_count()
>>> print(cnt)
1
>>> device = PyK4A(device_id=0)
>>> device.open()
[2021-04-10 16:15:38.616] [error] [t=18753] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (370): find_libusb_device(). Unable to open LIBUSB at index 0
[2021-04-10 16:15:38.618] [error] [t=18753] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2021-04-10 16:15:38.618] [error] [t=18753] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/pyk4a/pyk4a.py", line 68, in open
    self._device_open()
  File "/usr/local/lib/python3.6/dist-packages/pyk4a/pyk4a.py", line 85, in _device_open
    _verify_error(res)
  File "/usr/local/lib/python3.6/dist-packages/pyk4a/errors.py", line 18, in _verify_error
    raise K4AException()
pyk4a.errors.K4AException
```

Please let me know if you wish for more information.  This is likely an SDK issue, but if you have any clues I'd be really grateful.  Thanks so much!  Best regards.

lpasselin, 2021-04-10:
`find_libusb_device(). Unable to open LIBUSB at index 0`
We usually assume the device_id==0 but maybe that isn't the case on your system?

I am really not sure how to debug this since as you said, it is related to the SDK.

lpasselin, 2021-04-10:
Also, check this out. Maybe your problem is related to https://github.com/etiennedub/pyk4a/issues/93#issuecomment-751647738

michhar, 2021-04-10:
@lpasselin Thank you for your reply!  Interestingly, I can get to my Kinect DK RGB camera with OpenCV in the docker container (using `/dev/video1`) so believe the device can connect through usb and docker, but maybe it's something else.  I tried a `device_id=1`, but this did not work with above code sample.  I was using the Ubuntu packages `libk4a1.4`, `libk4a1.4-dev` and `k4a-tools`.

Now, I am building the Azure Kinect Sensor SDK from source and will let you know if it fixes this issue (using the `develop` branch actually which may have some Python bindings as well).  I am using `--privileged` to run my docker container btw which is how I can access it.  Stay tuned... :)

lpasselin, 2021-04-11:
My initial feeling was that this might be a permission error or bad idea rules.
But since you have it running with opencv, it doesn't make any sense. I really don't know how to help.

Is it possible you were root when running

I don't think you will have more luck by compiling the SDK.

lpasselin, 2021-04-11:
The SDK devs might be able to help you more than me unfortunately.
@shagren do you have an idea what might be the problem?

Also, if you are able to publish the dockerfile, we would be able to  confirm your system is not the problem

michhar, 2021-04-11:
Hi @lpasselin.  Thanks so much for your kind help.  Agree, seems like possibly a permission or udev rule error so I will check again on udev rules once my device can run.  It will take some time as I crashed my system building the SDK because forgot my swap file :(.  The device has just 4gb memory and I had only 2gb storage left.  It is running a version of RHEL and is an ARM64 device upon which I have my Ubuntu 18.04 docker container running.  I was root for both opencv and the pyk4a Python code snippet in that container.  I will publish the dockerfile as soon as I can.  Please let me know if you prefer I open an issue on the SDK project repo instead and thanks again.  :)

michhar, 2021-04-13:
Hi there.  I've placed my Dockerfile in a gist here:  https://gist.github.com/michhar/5d827fa6a05180e8c1eef33d0188720a.  Feel free to share any insights.  Thanks so much.  :)

shagren, 2021-04-15:
1. Please run docker with ` -v /dev/bus/usb:/dev/bus/usb  --privileged`
2. You can use kinect without dept camera:
```python
import pyk4a
config = pyk4a.Config(depth_mode=pyk4a.DepthMode.OFF, synchronized_images_only=False)
device = pyk4a.PyK4A(config=config)
device.open()
print(device.serial)
device.start()
capture = device.get_capture()
print(capture.color)
```
3. You cannot start depth camera, next error will be raised:
```
[2021-04-15 08:41:15.363] [error] [t=74] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (154): depth_engine_start_helper(). Depth engine create and initialize failed with error code: 204.
[2021-04-15 08:41:15.363] [error] [t=74] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (160): deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper()
[2021-04-15 08:41:15.363] [error] [t=74] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (194): depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread()
[2021-04-15 08:41:15.363] [error] [t=62] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (552): dewrapper_start(). Depth Engine thread failed to start
[2021-04-15 08:41:15.363] [error] [t=62] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (398): dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start()
[2021-04-15 08:41:15.363] [error] [t=62] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[2021-04-15 08:41:15.363] [error] [t=62] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[2021-04-15 08:41:15.363] [error] [t=62] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (895): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()
```
4.  I think problem with OpenGL which is required for Kinect Depthengine. For linux OpenGL usualy available via `X` server. You can play with `X ` server inside your container. 
5. Please share you results here  :)

michhar, 2021-04-26:
@shagren Thanks for the response!!  In fact, I am getting the same error as in original post above even with:

```
import pyk4a
config = pyk4a.Config(depth_mode=pyk4a.DepthMode.OFF, synchronized_images_only=False)
device = pyk4a.PyK4A(config=config)
device.open()
print(device.serial)
device.start()
capture = device.get_capture()
print(capture.color)
```

So, for me the color camera may not work, even, through `pyk4a` (although good with OpenCV).  I have installed the requirements from the [Linux dependencies](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#linux-dependencies) so in theory should have OpenGL and Kinect Depth engine.  Looks like my issue is before this check.

Any recommendations to try?  Should I try the Kinect Sensor SDK python bindings directly (build the SDK from source?).  Best Regards to you both.

lpasselin, 2021-05-19:
> Hi there. I've placed my Dockerfile in a gist here: https://gist.github.com/michhar/5d827fa6a05180e8c1eef33d0188720a. Feel free to share any insights. Thanks so much. :)

@michhar Did you progress with this issue?

The k4a libraries are missing from your dockerfile.
I assume you have a Dockerfile that installs libk4a from the official Microsoft package (or compiles from source).
Could you share that Dockerfile you are using?

lpasselin, 2021-05-19:
Closing the issue because it is probably unrelated to pyk4a.
But let's continue the discussion.

closed 2021-05-19 by lpasselin

michhar, 2021-05-21:
@lpasselin Thank you!  I had been installing the libk4a in the running docker container because I had, as you saw, not done so in the build - I was in testing mode.  I have another Dockerfile I will share, but want to ensure it builds correctly.  I greatly appreciate your time and effort as I agree this is more of a k4a issue.  Thanks again.