etiennedub/pyk4a #127: DLL load failed for k4a_module

tquintudelft-source, 2021-05-31:
Hi,

I'm currently getting this error when trying to run my python application:

```DLL load failed while importing k4a_module: The specified module could not be found.```

Both the /lib and /bin directories are in the PATH and pyk4a is installed in the environment. I've tried reinstalling the Azure SDK again and installing pyk4a in a new environment but no luck. A similar [issue](https://github.com/etiennedub/pyk4a/issues/23) is also on this repo, but no concrete solution has been posted.

Any help would be appreciated.

lpasselin, 2021-05-31:
Hi, check here.
https://github.com/etiennedub/pyk4a/issues/117#issuecomment-816816592

There are two solutions proposed in that thread.

tquintudelft-source, 2021-05-31:
> Hi, check here.
> [#117 (comment)](https://github.com/etiennedub/pyk4a/issues/117#issuecomment-816816592)
> 
> There are two solutions proposed in that thread.

Thanks, will take a look!

lpasselin, 2021-05-31:
We've had this problem for a while now. We need to make sure it never happens again.
I will make a PR adding a try except over the failing line and redirect users to a FAQ page or a specific part of the README.md explaining a solution. 

@shagren did you have to modify anything since your post [here](https://github.com/etiennedub/pyk4a/issues/117#issuecomment-816801446)?

shagren, 2021-05-31:
@lpasselin , I  will propose some changes in few days

closed 2021-06-01 by tquintudelft-source

qenops, 2021-07-21:
@shagren were these changes ever proposed/accepted? I've got some hacky code to load the dll by opening an MKV using open3d, which then enables pyk4a to be loaded. It looks like your branch might fix this, as well as simplifying the installation process.

shagren, 2021-07-22:
@qenops , only @lpasselin know when next version will be released :)

lpasselin, 2021-08-07:
New version 1.3.0 is released on pypi.
Support for windows automatic install was added. I hope it reduces our windows users problems!
Thanks again @shagren for the multiple PR. Your contribution is greatly appreciated!

Shihab-Shahriar, 2022-01-17:
Just wanted to note that I have this problem in version 1.3.0. I'm on Windows.