etiennedub/pyk4a #128: Misalignment of depth and video frames

tquintudelft-source, 2021-06-01:
I'm having this problem where the capture window for the lidar sensor and visual sensor are out of alignment:

[image]

I've worked with pyrealsense on the Intel L515 before and it has its own methods for frame alignment like so:

```
aligned_frames = self.align.process(frames)
color_frame = aligned_frames.get_color_frame()
 ```

Is there a similar solution in pyk4a or am I just doing something totally wrong?

shagren, 2021-06-01:
https://github.com/etiennedub/pyk4a/blob/8a2fb92f73067b97c35d82b337bc0d0920ffb5b7/example/viewer_transformation.py#L22

shagren, 2021-06-01:
you can use `capture.transformed_color` or `capture.transformed_depth`

closed 2021-06-01 by lpasselin