etiennedub/pyk4a #129: Saving only part of a capture with PyK4ARecord

chrismeunier, 2021-06-01:
I was testing the recording/playback functionalities you implemented in v1.2.0 (thanks for all you added recently by the way!) and I wondered if there would be some way to save only part of a capture.
I'm only interested in saving the `depth_point_cloud` and `transformed_color` arrays of the capture. I tried deleting the capture attributes I didn't need but that doesn't work (and doesn't seem a really good idea since technically I need the `color` and `depth` images to get what I want).

Previously I was simply adding the arrays I wanted to a list that I pickled in a binary file, this worked but steadily filled my RAM. Now with the recorder the RAM gets filled even faster since I save so much more data.
So is there maybe a trick or a workaround to avoid this? I don't care if the mkv file isn't readable with the Azure Kinect Viewer, only that I can recover the data with a `PyK4APlayback`.

Here's a piece of code corresponding to my workflow, just in case I'm doing something wrong...
I tried moving the `flush()` right after the `write_capture()` but the writing speed of my SSD is definitely too low to follow and I lose a *lot* of frames.

```python
from pyk4a import PyK4A, PyK4ARecord, connected_device_count
from pyk4a import Config, ColorResolution
from timeit import default_timer as timer

# Find Kinects
k4a_iterator = range(connected_device_count())
# Config for all default parameters except color resolution
config_1536p = Config(color_resolution=ColorResolution.RES_1536P)
# Initialize Kinects in a list
k4a_list = [PyK4A(config=config_1536p, device_id=i) for i in k4a_iterator]
for k4a in k4a_list:
    k4a.start()

# Create PyK4ARecord-ers
record_duration = 2 # 6-7 is max at 1536P with 3 cameras for 16Go of RAM
rec_filenames = ['data/' + k4a.serial + '.mkv' for k4a in k4a_list]
recorders = list()

for rec_name, config, k4a in zip(rec_filenames, config_list, k4a_list):
    recorders.append(PyK4ARecord(rec_name, config, k4a))
for k4a_rec in recorders:
    k4a_rec.create()

# Start capture saving
start_time = timer()
while timer() - start_time < record_duration:
    for k4a, k4a_rec in zip(k4a_list, recorders):
        capture = k4a.get_capture()
        # ? here save only what I need ?
        k4a_rec.write_capture(capture)

# Print how many captures were saved by camera
print('Captures saved by Kinect:')
print([f"{i}: {k4a_rec.captures_count}" for i, k4a_rec in enumerate(recorders)])

# Flushing and closing recorders
for k4a_rec in recorders:
    k4a_rec.flush()
    k4a_rec.close()

# Stopping the Kinects
for k4a in k4a_list:
    k4a.stop()
```
NB: I'm only using the 1536p resolution for the 4:3 aspect ratio to get a complete overlap with the depth sensor.

shagren, 2021-06-01:
1. `depth_point_cloud` is restored from depth image which are smaller than point cloud self.

2. Try to use compressed color images. Please add `color_format=ImageFormat.COLOR_MJPG` to `Config(...)`. In this case color image will be much smaller. 
You can play with different formats, please check https://github.com/etiennedub/pyk4a/blob/master/example/color_formats.py. 
Maybe you need decompress color image in some cases, you can use something like:

 ```
   # will work for mjpeg
   color_image = cv2.imdecode(capture.color, cv2.IMREAD_UNCHANGED)
   rgba_image = cv2.cvtColor(color_image, cv2.COLOR_RGB2RGBA)
```
You need decompressed image for `color_image_to_depth_camera()` transformation

3. Use sdd disks

shagren, 2021-06-01:
BTW: we use 720P for space/bandwith saving

chrismeunier, 2021-06-04:
Anyway the color format change to MJPG solves my problem so I will close the issue, thanks for your ideas!

closed 2021-06-04 by chrismeunier