etiennedub/pyk4a #131: How to use IR fill light

zhaocheng-hub, 2021-06-04:
Hello, I want to use IR lighting,
How to set the RESCALE_IR_TO_MONO8 parameter value of Kinect sensor?

lpasselin, 2021-06-04:
Hi, we do not support this parameter yet.

lpasselin, 2021-06-04:
Also this parameter is not in the SDK docs... Could you elaborate?

zhaocheng-hub, 2021-06-04:
hello
I used it in the link, [https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/docs/usage.md](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/docs/usage.md) 

`Ir_mono8_scaling_factor (float)` :Defaults to 1.0. Scaling factor to apply when converting IR to mono8 (see rescale_ir_to_mono8). If using Illumination,Use the value 0.5-1. If using passive IR, use 10.

Thank you for your。

lpasselin, 2021-06-04:
Can this be done by simply multiplying the capture.ir inplace by a scalar?
`capture.ir *= 10`

I'll have to look into the ros docs to understand what they do exactly.

zhaocheng-hub, 2021-06-05:
Thank you for your reply.
I want to use infrared lighting.

lpasselin, 2021-06-15:
pyk4a does not support this feature since it is a trivial operation that you can perform yourself after retrieving the image.