etiennedub/pyk4a #133: Problem using k4a_module.transformation_depth_image_to_color_camera_custom

msqiu, 2021-06-27:
I tried to transform the IR images but I found that if the corresponding depth pixels are missing, the IR pixels will be 0.
I read the original Azure Kinect DK code and found that they add a parameter [**invalid_custom_value**:](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_gac00dd00e7612a86382e3d0a130f276bb.html#gac00dd00e7612a86382e3d0a130f276bb)
<html>
<body>

invalid_custom_value | Defines the custom image pixel value that should be written to transformed_custom_image in case the corresponding depth pixel can not be transformed into the color camera space.
-- | --

</body>
</html>

And this is missing here:
```
k4a_module.transformation_depth_image_to_color_camera_custom(
        calibration.transformation_handle, thread_safe, depth, custom, calibration.color_resolution, interp_nearest,
    )
```

I hope you can solve this issue. Thanks!

lpasselin, 2021-06-28:
Hi, is this still an issue?

msqiu, 2021-06-28:
No, you can close the issue now:)

closed 2021-06-28 by msqiu