etiennedub/pyk4a #134: rgb, ir, depth recording method

kimtaehyeong, 2021-07-06:
Hello.

I'm writing code to record via kinect.
Currently, I am trying to record ir, rgb, and depth images.
https://github.com/etiennedub/pyk4a/blob/master/example/record.py I am referring to the code here,
How can I record if I want to record three at the same time other than rgb?
Thank you.

lpasselin, 2021-07-10:
Hi, can you share code showing what you try to make work?

kimtaehyeong, 2021-07-10:
Currently, I save as an image through the cv.imwrite function of the opencv function (the image is too large), but the cost is too high, so I want to save it as a video.

The previous code is currently being written by the company, so it is difficult to share, and the temporary code is as follows.
```
count_time = 10
img_count = 1

def main():
    
    global img_count

    config_1 = Config(wired_sync_mode=0, camera_fps=1, color_resolution=pyk4a.ColorResolution.RES_3072P, depth_mode=pyk4a.DepthMode.WFOV_UNBINNED,)
    k4a_1 = PyK4A(device_id=0, config=config_1)
    k4a_1.start()

    color = []
    ir = []
    depth = []
    t_start = time.time()

    while True:
        capture = k4a_1.get_capture()
        key = cv2.waitKey(10)
        # if capture.color is not None:
        #     cv2.imshow("Color", cv2.resize(capture_1.color, dsize=(640, 480), interpolation=cv2.INTER_AREA))
        #     cv2.imshow("Transformed IR", colorize(capture_1.ir, (None, 500), colormap=cv2.COLORMAP_JET))
        if key == ord('s'):
            img_count = img_count + 1
            print(img_count)
            color.append(capture.color)
            ir.append(capture.ir)
            depth.append(capture.depth)   

            if img_count > 5: break
        t_end = time.time()
  

    print(f"time elapsed: {t_end - t_start}")
    print(f"n frames: {len(color1)}")
    t_start = t_end
    
    k4a_1.stop()

    for i in range(len(ir)):
        cv2.imwrite('save_img_1/transformed_color0-{}.jpg'.format(str(i)),color[i])
        cv2.imwrite('save_img_1/transformed_depth0-{}.png'.format(str(i)), onechannel_to_rgb, depth[i])
        cv2.imwrite('save_img_1/transformed_ir0-{}.png'.format(str(i)), onechannel_to_rgb, ir2[i])

    t_end = time.time()
    
    fps = (t_end - t_start) / len(color1)
    print(f"time elapsed: {t_end - t_start},  {fps}")

if __name__ == "__main__":

    main()
```

The method I want is to save the video taken for 10 seconds as a video. Currently, when the s button is pressed, it is saved for 5 frames.

Thank you.

lpasselin, 2021-07-13:
Sorry, I am not sure what you are asking.

You can use the recording feature to record more than a single frame...
Try to record captures and playback the recording to see if it works for you.
Try the examples.

lpasselin, 2021-07-22:
Reopen if required

closed 2021-07-22 by lpasselin