etiennedub/pyk4a #136: How about point cloud?

wangmiaowei, 2021-07-15:
Is there any way to get point cloud from Azure Kinect .mkv file?

chrismeunier, 2021-07-15:
Sure you can use the code in [example/playback.py](https://github.com/etiennedub/pyk4a/blob/8a2fb92f73067b97c35d82b337bc0d0920ffb5b7/example/playback.py#L16) and recover the point cloud with `capture.depth_point_cloud`.

Simplest way to test it I can think of:
```python
from pyk4a import PyK4APlayback

playback= PyK4APlayback('your_path_to_mkv_file')
playback.open()

while True:
    try:
        capture = playback.get_next_capture()
        if capture.depth_point_cloud is not None:
            point_cloud = capture.depth_point_cloud
            # then do what you want
    except EOFError:
        break
```

wangmiaowei, 2021-07-15:
Thanks for your suggestion. I'd like to try it.

closed 2021-07-15 by lpasselin

wangmiaowei, 2021-07-15:
@lpasselin  Is there any function to save the points file?

lpasselin, 2021-07-15:
Any method to save a numpy array will work.

wangmiaowei, 2021-07-17:
Thanks for your reply. Is there any function in this python wrap to transform the depth image to PointCloud?