etiennedub/pyk4a #142: Cannot import k4a_module

fatimahfirdauss, 2021-08-12:
Hi,
I use windows and I pip install pyk4a through my cmd and it is successfully installed, below is the message that I got.
Processing c:\users\ayeuser4\appdata\local\pip\cache\wheels\53\5f\d6\2578c50004084994a636f0124ba027261e51614a0ee759aa73\pyk4a-1.3.0-cp38-cp38-win_amd64.whl
Requirement already satisfied: python-version>="3.4" in c:\users\ayeuser4\appdata\local\programs\python\python38\lib\site-packages (from pyk4a) (0.0.2)
Requirement already satisfied: numpy in c:\users\ayeuser4\appdata\local\programs\python\python38\lib\site-packages (from pyk4a) (1.19.5)
Installing collected packages: pyk4a
Successfully installed pyk4a-1.3.0

When I tried to run the basic example displaying the first frame that you gave to test out, I got this error:
from pyk4a import PyK4A
ModuleNotFoundError: No module named 'pyk4a'

so I installed pyk4a again through pycharm python interpreter and run the code again and I get this error:
ImportError: Cannot import k4a_module. DLL directory added was C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\AMD64\release\bin. You can provide a different path containing `k4a.dll`using the environment variable `K4A_DLL_DIR`. Also make sure pyk4a was properly built.

I'm a beginner and very clueless on how to fix this, can you help please?

fatimahfirdauss, 2021-08-12:
I pip uninstall pyk4a and install again using 
pip install pyk4a --no-use-pep517 --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"

Run the same code and got the same error
ImportError: Cannot import k4a_module. DLL directory added was C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\AMD64\release\bin. You can provide a different path containing `k4a.dll`using the environment variable `K4A_DLL_DIR`. Also make sure pyk4a was properly built.

shagren, 2021-08-12:
@fatimahfirdauss , do you use 64-bit python?

fatimahfirdauss, 2021-08-13:
64 bit

shagren, 2021-08-17:
I have no idea :(
I use  3.8 , 64bit on win10 and haven't any problem.

lpasselin, 2021-08-23:
Can you make sure the directory `C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\AMD64\release\bin` contains k4a.dll ?

l-06, 2021-08-30:
Hi, I had the same problem as you, my classmate helped me, you may not have added the SDK path to the system environment variable

lpasselin, 2021-08-31:
@fatimahfirdauss 
Normally the SDK path is supposed to be automatically detected since 1.3.0. If that does not work, you can try specifying the env variable `K4A_DLL_DIR` with the path of the k4a.dll. This path normally looks like `C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\AMD64\release\bin`. In your case, that path is probably wrong since the automatic dll loader finds that path and yet cannot load it.

Any other information to help us debug this?

AndreSouto, 2021-09-01:
@fatimahfirdauss 
I had exactly the same problem. I was using python 3.7. When I updated to 3.8 and reinstalled everything, it worked fine.

lpasselin, 2021-09-28:
@fatimahfirdauss is this still an issue?

ShaoxiongYao, 2021-10-02:
Actually I had a similar issue before, and it was caused by my `numpy` too old. I used `pip install numpy --update` to fix it.

closed 2021-10-04 by lpasselin

7oud, 2021-12-09:
the same question, I set env variable, still not work, I have no idea

SeanHRI, 2021-12-12:
I met the same problem and tried many ways like using different python versions, but they did not work. Finally, after adding the path containing the k4a.dll to the path of the system variable, and reboot my computer, this problem is no longer existing.

YoniChechik, 2022-03-06:
In windows: adding `C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\bin` to the PATH env var (via `edit envinronment variables`) solved it for me

owen2t, 2022-04-15:
I'm having the same problem. I was able to use pyk4a yesterday and updated my pc (windows 10 version ? to windows 10 version21H2). I added `C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\bin` to the PATH and restarted VScode and my pc with no success. I also checked that k4a.dll was in the folder.

7oud, 2022-04-29:
@owen2t I tried it again, re-installed numpy, still not work. /sad

Carlos-biru, 2022-05-06:
@owen2t Worked for me, but I didn't have to update numpy.
numpy version is 1.21.5
Python version is 3.8.13

After trying everything above it solved the issue adding the path to the "PATH" env var.
thanks! :)

rajkundu, 2022-05-27:
I had this issue as well, and like @Carlos-biru, I added `C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\AMD64\release\bin` to `PATH` to get pyk4a to import successfully. Any idea why it can't detect the DLL itself, which is installed in the default location?

Wooxinyu, 2024-04-30:
I had added C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\AMD64\release\bin to PATH，and k4a.dll in this path,
but ImportError: Cannot import k4a_module. DLL directory added was C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\bin