etiennedub/pyk4a #144: pip install issue

PingHsunTsai, 2021-08-30:
Hi there, 
Recently I run into this issue while doing  `pip install pyk4a --no-use-pep517 --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib""` with Azure Kinect SDK 1.4.1 installed.

following is the error message:
`C:\Users\PingHsunTsai\anaconda3\envs\kinect_dk\lib\site-packages\pip\_internal\commands\install.py:229: UserWarning: Disabling all use of wheels due to the use of --build-option / --global-option / --install-option.
  cmdoptions.check_install_build_global(options)
Collecting pyk4a
  Using cached pyk4a-1.3.0.tar.gz (23 kB)
Collecting numpy
  Using cached numpy-1.21.2.zip (10.3 MB)
Requirement already satisfied: python_version>="3.4" in c:\users\pinghsuntsai\anaconda3\envs\kinect_dk\lib\site-packages (from pyk4a) (0.0.2)
Skipping wheel build for pyk4a, due to binaries being disabled for it.
Skipping wheel build for numpy, due to binaries being disabled for it.
Installing collected packages: numpy, pyk4a
    Running setup.py install for numpy ... error
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\PingHsunTsai\anaconda3\envs\kinect_dk\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\PingHsunTsai\\AppData\\Local\\Temp\\pip-install-h0uuembw\\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\\setup.py'"'"'; __file__='"'"'C:\\Users\\PingHsunTsai\\AppData\\Local\\Temp\\pip-install-h0uuembw\\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' build_ext '-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include' '-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"' install --record 'C:\Users\PingHsunTsai\AppData\Local\Temp\pip-record-k7e34oh9\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\PingHsunTsai\anaconda3\envs\kinect_dk\Include\numpy'
         cwd: C:\Users\PingHsunTsai\AppData\Local\Temp\pip-install-h0uuembw\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\
    Complete output (32 lines):
    Running from numpy source directory.
    Cythonizing sources
    Processing numpy/random\_bounded_integers.pxd.in
    Processing numpy/random\bit_generator.pyx
    Traceback (most recent call last):
      File "C:\Users\PingHsunTsai\AppData\Local\Temp\pip-install-h0uuembw\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\tools\cythonize.py", line 59, in process_pyx
        import Cython
    ModuleNotFoundError: No module named 'Cython'

    The above exception was the direct cause of the following exception:

    Traceback (most recent call last):
      File "C:\Users\PingHsunTsai\AppData\Local\Temp\pip-install-h0uuembw\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\tools\cythonize.py", line 240, in <module>
        main()
      File "C:\Users\PingHsunTsai\AppData\Local\Temp\pip-install-h0uuembw\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\tools\cythonize.py", line 236, in main
        find_process_files(root_dir)
      File "C:\Users\PingHsunTsai\AppData\Local\Temp\pip-install-h0uuembw\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\tools\cythonize.py", line 227, in find_process_files
        process(root_dir, fromfile, tofile, function, hash_db)
      File "C:\Users\PingHsunTsai\AppData\Local\Temp\pip-install-h0uuembw\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\tools\cythonize.py", line 193, in process
        processor_function(fromfile, tofile)
      File "C:\Users\PingHsunTsai\AppData\Local\Temp\pip-install-h0uuembw\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\tools\cythonize.py", line 66, in process_pyx
        raise OSError(msg) from e
    OSError: Cython needs to be installed in Python as a module
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\PingHsunTsai\AppData\Local\Temp\pip-install-h0uuembw\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\setup.py", line 448, in <module>
        setup_package()
      File "C:\Users\PingHsunTsai\AppData\Local\Temp\pip-install-h0uuembw\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\setup.py", line 430, in setup_package
        generate_cython()
      File "C:\Users\PingHsunTsai\AppData\Local\Temp\pip-install-h0uuembw\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\setup.py", line 236, in generate_cython
        raise RuntimeError("Running cythonize failed!")
    RuntimeError: Running cythonize failed!
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\PingHsunTsai\anaconda3\envs\kinect_dk\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\PingHsunTsai\\AppData\\Local\\Temp\\pip-install-h0uuembw\\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\\setup.py'"'"'; __file__='"'"'C:\\Users\\PingHsunTsai\\AppData\\Local\\Temp\\pip-install-h0uuembw\\numpy_ff1884afbd734bbb9d24d4cfa04d6ad4\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' build_ext '-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include' '-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"' install --record 'C:\Users\PingHsunTsai\AppData\Local\Temp\pip-record-k7e34oh9\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\PingHsunTsai\anaconda3\envs\kinect_dk\Include\numpy' Check the logs for full command output.`

any ideas?
Thanks in advance!

closed 2021-08-31 by PingHsunTsai

lpasselin, 2021-08-31:
Sorry I saw too late. What was the solution?

PingHsunTsai, 2021-08-31:
Hey @lpasselin,
No worry! I didn't solve it actually, but I'm checking another [package ](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/src/python/k4a)that provides similar use. 
But I am still open to use this package if someone can have a look at the error.

lpasselin, 2021-08-31:
Honestly it's the first time that I hear about official python bindings. I am happy that Microsoft is doing that!

I am not sure but I think the error above displays numpy installation error. Can you confirm `pip install numpy works`? Then try again `pip install pyk4a`.