etiennedub/pyk4a #145: Unhandled exception in script error

euijaebilly, 2021-09-10:
i try to make exe file with azure kinect
there are occured this error message

[image]

how can i solve this problem?

lpasselin, 2021-09-10:
There should be logs in a console. K4AExceptions are usually verbose.

Without more information there is nothing I can do. Please provide more information if you want more help.

Maybe you should practice using pyk4a in a small script. See the examples provided.

euijaebilly, 2021-09-10:
[image]

euijaebilly, 2021-09-10:
i try run tiny code for test 
`
import cv2
import pyAzKinct

camera1 = pyAzKinct.transfer_img(0, "master")

cimg1,dimg1,_ = pyAzKinct.get_images(camera1)  # get image1 from camera1

cv2.imshow("test1",cimg1)

cv2.waitKey()
`
and has same error.

euijaebilly, 2021-09-10:
내가 테스트를위한 작은 코드를 실행하려고
`
수입 CV2의
수입 pyAzKinct을

카메라1 = pyAzKinct.transfer_img(0, "마스터")

cimg1,dimg1,_ = pyAzKinct.get_images(camera1) # 카메라1에서 이미지1 가져오기

cv2.imshow("test1",cimg1)

cv2.waitKey()
`
및 동일한 오류가 있습니다.

lpasselin, 2021-09-10:
PyAzKinct is not the library in this repository... We develop pyl4a

lpasselin, 2021-09-10:
But still, read the error code. You must install the depthengine properly.

euijaebilly, 2021-09-23:
well pyAzKinct is just a python script for using some features of pyk4a.
So, basically, erros are occurring in pyk4a, and if it is caused by depthEngine, how can I install it properly?

lpasselin, 2021-09-23:
Can you share pyAzKinct source code?

euijaebilly, 2021-09-23:
`# Path to the module
# TODO: Modify with the path containing the k4a.dll from the Azure Kinect SDK
modulePath = 'C:\\Program Files\\Azure Kinect SDK v1.4.1\\sdk\\windows-desktop\\AMD64\\release\\bin\\k4a.dll'

import cv2
import numpy as np
from typing import Optional, Tuple
import pyk4a
from pyk4a import *
import psutil

def memory_usage(message: str = 'debug'):
    # current process RAM usage
    p = psutil.Process()
    rss = p.memory_info().rss / 2 ** 20 # Bytes to MB
    print(f"[{message}] memory usage: {rss: 10.5f} MB")

##############################################################
# name:device_master_config
#
# parameter: camera   output: config
#
# description: this method setting camera with a config for master
#
##############################################################
def device_master_config():
    cfg = Config(
            color_format=pyk4a.ImageFormat.COLOR_BGRA32,
            color_resolution=pyk4a.ColorResolution.RES_1080P,
            depth_mode=pyk4a.DepthMode.WFOV_UNBINNED,
            camera_fps=FPS.FPS_5,
            synchronized_images_only=True,
            wired_sync_mode=pyk4a.WiredSyncMode.MASTER
        )

    return cfg

##############################################################
# name: device_subordinate_config
#
# parameter: camera    output: config
#
# description:this method setting camera with a config for slave
#
##############################################################
def device_subordinate_config():
    cfg = Config(
            color_format=pyk4a.ImageFormat.COLOR_BGRA32,
            color_resolution=pyk4a.ColorResolution.RES_1080P,
            depth_mode=pyk4a.DepthMode.WFOV_UNBINNED,
            camera_fps=FPS.FPS_5,
            synchronized_images_only=True,
            wired_sync_mode=pyk4a.WiredSyncMode.SUBORDINATE
        )
    return cfg

def colorize(
        image: np.ndarray,
        clipping_range: Tuple[Optional[int], Optional[int]] = (None, None),
        colormap: int = cv2.COLORMAP_HSV,
) -> np.ndarray:
    if clipping_range[0] or clipping_range[1]:
        img = image.clip(clipping_range[0], clipping_range[1])
    else:
        img = image.copy()
    img = cv2.normalize(img, None, 0, 255, cv2.NORM_MINMAX, dtype=cv2.CV_8U)
    img = cv2.applyColorMap(img, colormap)
    return img

##############################################################
# name:get_images
#
# parameter: camera object    output: RGB-D image
#
# description: get RGB and depth image from parameter camera
# and return RGB-D image
#
##############################################################
def get_images(camera):
    # Get capture
    capture=camera.get_capture()

    transformed_depth = cv2.normalize(capture.transformed_depth, None, 0, 255, cv2.NORM_MINMAX, dtype=cv2.CV_8U)

    transformed_depth_color_image =cv2.applyColorMap(np.round(capture.transformed_depth / 30).astype(np.uint8),
                  cv2.COLORMAP_JET)
    comb_img = cv2.addWeighted(capture.color[:,:,:3], 0.7,transformed_depth_color_image, 0.3, 0)
    #cv2.imshow("Transformed Color", comb_img)
    return capture.color[:,:,:3],transformed_depth,comb_img

##############################################################
# name: transfer_img
#
# parameter: index of camera, config(master or slave)
# output: camera object
#
# description: return camera object with setting config
#
##############################################################
def transfer_img(camera_idx, cfg):
    if cfg == "master":
        camera_cfg = device_master_config()
    elif cfg == "sub":
        camera_cfg = device_subordinate_config()

    camera = PyK4A(device_id=camera_idx, config=camera_cfg)
    camera.start()

    return camera
`

just simple code.

lpasselin, 2021-09-23:
Did you try to figure out why the depthengine is not found?

euijaebilly, 2021-09-23:
Yes. I tried finding depthEngine_2_0.dll from "C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\x86\release\bin" 
and placing it in the same path as the .exe file and running it, 
or adding to the environment variable, but it doesn't work.

euijaebilly, 2021-09-24:
> Did you try to figure out why the depthengine is not found?

Yes. I tried finding depthEngine_2_0.dll from "C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\x86\release\bin"
and placing it in the same path as the .exe file and running it,
or adding to the environment variable, but it doesn't work.

There were no problems when running in Python. My guess is that it doesn't find the correct path as it converts to .exe.

euijaebilly, 2021-09-28:
Did you find any solution?

You are seem really busy now.....

lpasselin, 2021-09-28:
I have no idea how .exe work in python.
For example see here there are specific instructions to add .dll with pyinstaller.
https://pyinstaller.readthedocs.io/en/stable/spec-files.html#adding-data-files

This is not really a pyk4a problem. Maybe you should look into the library used to create your .exe. How do they handle the .dll ?

Otherwise you could look into adding the depthengine dll manually: https://github.com/etiennedub/pyk4a/blob/master/pyk4a/win32_utils.py#L11

lpasselin, 2021-10-04:
I'll close as this is most likely an error caused by bad configuration of the executable generating tool.

closed 2021-10-04 by lpasselin