etiennedub/pyk4a #146: errors occurred in the pip

Qiu731, 2021-09-28:
Hi! I run pip install pyk4a --no-use-pep517 --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"" 
But errors have occurred, I hope you can help me solve this problem. Thank you.
And the error is as follows:   

ERROR: Command errored out with exit status 1:
     command: 'D:\Anaconda\envs\pykinz\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\小Q\\AppData\\Local\\Temp\\pip-install-yjzlj7wt\\pyk4a_d65eb862730e4846bcd089d1253096d6\\setup.py'"'"'; __file__='"'"'C:\\Users\\小Q\\AppData\\Local\\Temp\\pip-install-yjzlj7wt\\pyk4a_d65eb862730e4846bcd089d1253096d6\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' build_ext '-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include' '-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"' install --record 'C:\Users\小Q\AppData\Local\Temp\pip-record-1dr4lq_t\install-record.txt' --single-version-externally-managed --compile --install-headers 'D:\Anaconda\envs\pykinz\Include\pyk4a'
         cwd: C:\Users\小Q\AppData\Local\Temp\pip-install-yjzlj7wt\pyk4a_d65eb862730e4846bcd089d1253096d6\
    Complete output (20 lines):
    Automatically detected kinect SDK. Adding include dir: C:\Program Files\Azure Kinect SDK v1.4.1\sdk\include and library dir C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\AMD64\release\lib.
    D:\Anaconda\envs\pykinz\lib\site-packages\setuptools\dist.py:717: UserWarning: Usage of dash-separated 'description-file' will not be supported in future versions. Please use the underscore name 'description_file' instead
      warnings.warn(
    running build_ext
    building 'k4a_module' extension
    creating build
    creating build\temp.win-amd64-3.8
    creating build\temp.win-amd64-3.8\Release
    creating build\temp.win-amd64-3.8\Release\pyk4a
    D:\visual studio\VC\Tools\MSVC\14.29.30037\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD "-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" -ID:\Anaconda\envs\pykinz\lib\site-packages\numpy\core\include "-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" -ID:\Anaconda\envs\pykinz\include -ID:\Anaconda\envs\pykinz\include "-ID:\visual studio\VC\Tools\MSVC\14.29.30037\ATLMFC\include" "-ID:\visual studio\VC\Tools\MSVC\14.29.30037\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-ID:\Windows Kits\10\include\10.0.19041.0\ucrt" "-ID:\Windows Kits\10\include\10.0.19041.0\shared" "-ID:\Windows Kits\10\include\10.0.19041.0\um" "-ID:\Windows Kits\10\include\10.0.19041.0\winrt" "-ID:\Windows Kits\10\include\10.0.19041.0\cppwinrt" /EHsc /Tppyk4a/pyk4a.cpp /Fobuild\temp.win-amd64-3.8\Release\pyk4a/pyk4a.obj
    pyk4a.cpp
    D:\Anaconda\envs\pykinz\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(14) : Warning Msg: Using deprecated NumPy API, disable it with #define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
    pyk4a/pyk4a.cpp(351): warning C4244: “参数”: 从“__int64”转换到“int32_t”，可能丢失数据
    pyk4a/pyk4a.cpp(370): warning C4244: “参数”: 从“__int64”转换到“int32_t”，可能丢失数据
    pyk4a/pyk4a.cpp(546): warning C4244: “初始化”: 从“npy_intp”转换到“int”，可能丢失数据
    pyk4a/pyk4a.cpp(547): warning C4244: “初始化”: 从“npy_intp”转换到“int”，可能丢失数据
    creating C:\Users\小Q\AppData\Local\Temp\pip-install-yjzlj7wt\pyk4a_d65eb862730e4846bcd089d1253096d6\build\lib.win-amd64-3.8
    D:\visual studio\VC\Tools\MSVC\14.29.30037\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO "/LIBPATH:C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\AMD64\release\lib" "/LIBPATH:C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"" /LIBPATH:D:\Anaconda\envs\pykinz\libs /LIBPATH:D:\Anaconda\envs\pykinz\PCbuild\amd64 "/LIBPATH:D:\visual studio\VC\Tools\MSVC\14.29.30037\ATLMFC\lib\x64" "/LIBPATH:D:\visual studio\VC\Tools\MSVC\14.29.30037\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\lib\um\x64" "/LIBPATH:D:\Windows Kits\10\lib\10.0.19041.0\ucrt\x64" "/LIBPATH:D:\Windows Kits\10\lib\10.0.19041.0\um\x64" k4a.lib k4arecord.lib /EXPORT:PyInit_k4a_module build\temp.win-amd64-3.8\Release\pyk4a/pyk4a.obj /OUT:build\lib.win-amd64-3.8\k4a_module.cp38-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.8\Release\pyk4a\k4a_module.cp38-win_amd64.lib
    LINK : fatal error LNK1104: 无法打开文件“studio\VC\Tools\MSVC\14.29.30037\ATLMFC\lib\x64 \LIBPATH:D:\visual.obj”
    error: command 'D:\\visual studio\\VC\\Tools\\MSVC\\14.29.30037\\bin\\HostX86\\x64\\link.exe' failed with exit status 1104
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'D:\Anaconda\envs\pykinz\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\小Q\\AppData\\Local\\Temp\\pip-install-yjzlj7wt\\pyk4a_d65eb862730e4846bcd089d1253096d6\\setup.py'"'"'; __file__='"'"'C:\\Users\\小Q\\AppData\\Local\\Temp\\pip-install-yjzlj7wt\\pyk4a_d65eb862730e4846bcd089d1253096d6\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' build_ext '-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include' '-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"' install --record 'C:\Users\小Q\AppData\Local\Temp\pip-record-1dr4lq_t\install-record.txt' --single-version-externally-managed --compile --install-headers 'D:\Anaconda\envs\pykinz\Include\pyk4a' Check the logs for full command output.

lpasselin, 2021-09-28:
Hi, did you try the basic `pip install pyk4a` ?

Qiu731, 2021-09-28:
> Hi, did you try the basic `pip install pyk4a` ?
> 
> Edit: Exit status 1104 see causes here: https://docs.microsoft.com/en-us/cpp/error-messages/tool-errors/linker-tools-error-lnk1104

It will return this error,but i have installed setuptools==58.1.0.
ERROR: Exception:
Traceback (most recent call last):
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\cli\base_command.py", line 173, in _main
    status = self.run(options, args)
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\cli\req_command.py", line 203, in wrapper
    return func(self, options, args)
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\commands\install.py", line 315, in run
    requirement_set = resolver.resolve(
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\resolution\resolvelib\resolver.py", line 94, in resolve
    result = self._result = resolver.resolve(
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 472, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 341, in resolve
    self._add_to_criteria(self.state.criteria, r, parent=None)
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 172, in _add_to_criteria
    if not criterion.candidates:
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_vendor\resolvelib\structs.py", line 151, in __bool__
    return bool(self._sequence)
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 140, in __bool__
    return any(self)
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 128, in <genexpr>
    return (c for c in iterator if id(c) not in self._incompatible_ids)
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 32, in _iter_built
    candidate = func()
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\resolution\resolvelib\factory.py", line 204, in _make_candidate_from_link
    self._link_candidate_cache[link] = LinkCandidate(
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\resolution\resolvelib\candidates.py", line 295, in __init__
    super().__init__(
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\resolution\resolvelib\candidates.py", line 156, in __init__
    self.dist = self._prepare()
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\resolution\resolvelib\candidates.py", line 227, in _prepare
    dist = self._prepare_distribution()
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\resolution\resolvelib\candidates.py", line 305, in _prepare_distribution
    return self._factory.preparer.prepare_linked_requirement(
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\operations\prepare.py", line 508, in prepare_linked_requirement
    return self._prepare_linked_requirement(req, parallel_builds)
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\operations\prepare.py", line 570, in _prepare_linked_requirement
    dist = _get_prepared_distribution(
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\operations\prepare.py", line 60, in _get_prepared_distribution
    abstract_dist.prepare_distribution_metadata(finder, build_isolation)
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\distributions\sdist.py", line 34, in prepare_distribution_metadata
    self._setup_isolation(finder)
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_internal\distributions\sdist.py", line 88, in _setup_isolation
    reqs = backend.get_requires_for_build_wheel()
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_vendor\pep517\wrappers.py", line 168, in get_requires_for_build_wheel
    return self._call_hook('get_requires_for_build_wheel', {
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_vendor\pep517\wrappers.py", line 328, in _call_hook
    raise BackendUnavailable(data.get('traceback', ''))
pip._vendor.pep517.wrappers.BackendUnavailable: Traceback (most recent call last):
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 89, in _build_backend
    obj = import_module(mod_path)
  File "D:\Anaconda\envs\pykinz\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'setuptools'

Qiu731, 2021-09-29:
> Hi, did you try the basic `pip install pyk4a` ?
> 
> Edit: Exit status 1104 see causes here: https://docs.microsoft.com/en-us/cpp/error-messages/tool-errors/linker-tools-error-lnk1104

Hi, I ran pip install pyk4a  --no-use-pep517 and it looks correct,but when I ran viewer.py , it made errorget.

**my install return is as follow:**
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Collecting pyk4a
  Using cached https://pypi.tuna.tsinghua.edu.cn/packages/cb/62/0888bf0fe47922d1033e12fdf966626c68b92abaf35b3770b994f5882bd9/pyk4a-1.3.0.tar.gz (23 kB)
Requirement already satisfied: numpy in d:\anaconda\envs\pykinz\lib\site-packages (from pyk4a) (1.21.2)
Requirement already satisfied: python_version>="3.4" in d:\anaconda\envs\pykinz\lib\site-packages (from pyk4a) (0.0.2)
Building wheels for collected packages: pyk4a
  Building wheel for pyk4a (setup.py) ... done
  Created wheel for pyk4a: filename=pyk4a-1.3.0-cp38-cp38-win_amd64.whl size=33419 sha256=a31d0517b9efff3c2a5a443622b8f4cec3274ef219a014457ebff825f4ae2dc8
  Stored in directory: c:\users\小q\appdata\local\pip\cache\wheels\4e\7f\db\9a9fbb5670b41c70e88d814cad1866476df79586eb1e575d0a
Successfully built pyk4a
Installing collected packages: pyk4a
Successfully installed pyk4a-1.3.0

**my error in running viewer.py is as follow:**
Traceback (most recent call last):
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pyk4a\module.py", line 5, in <module>
    import k4a_module  # noqa: F401
ImportError: DLL load failed while importing k4a_module: 找不到指定的模块。

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "viewer.py", line 4, in <module>
    import pyk4a
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pyk4a\__init__.py", line 1, in <module>
    from .calibration import Calibration, CalibrationType
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pyk4a\calibration.py", line 8, in <module>
    from .module import k4a_module
  File "D:\Anaconda\envs\pykinz\lib\site-packages\pyk4a\module.py", line 14, in <module>
    raise ImportError(
ImportError: Cannot import k4a_module. DLL directory added was C:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\bin\k4a.dll. You can provide a different path containing `k4a.dll`using the environment variable `K4A_DLL_DIR`. Also make sure pyk4a was properly built.

lpasselin, 2021-10-04:
The problem is that the kinect azure k4a.dll file is not found. See other issues related to `k4a.dll` or `DLL load failed while importing k4a_module` or `Cannot import k4a_module`. There are many of them.

You seem to be using disk D:\\. The automatic k4a.dll finder only looks in C:\\.
This means you might not have installed the kinect SDK library in the C:\\ folder. If that is the case you have to specify the SDK path during the install.

closed 2021-10-04 by lpasselin

Qiu731, 2021-10-10:
> The problem is that the kinect azure k4a.dll file is not found. See other issues related to `k4a.dll` or `DLL load failed while importing k4a_module` or `Cannot import k4a_module`. There are many of [...]

I am sorry to bother you.I have checked all the answers about k4a.dll but it doesn't work.And I have installed the kinect SDK library in the C:\ folder:
[image]