etiennedub/pyk4a #147: Segmentation fault with capture.transformed_depth

ekolve, 2021-09-30:
When I run the following snippet, a segfault is triggered and the machine the Azure Kinect is connected to crashes (shutdown).

```python
k4a = PyK4A(Config(color_resolution=pyk4a.ColorResolution.RES_720P, depth_mode=pyk4a.DepthMode.NFOV_UNBINNED,synchronized_images_only=True))
k4a.start()
capture = k4a.get_capture()
converted_image = capture.transformed_depth
k4a.stop()
```

Error:
```
[2021-09-30 10:59:32.155] [error] [t=27670] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    733544 type:Color
[2021-09-30 10:59:32.185] [error] [t=27670] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    766855 type:Color
[2021-09-30 10:59:32.217] [error] [t=27670] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    800188 type:Color
[2021-09-30 10:59:32.249] [error] [t=27670] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    833522 type:Color

[2021-09-30 10:59:32.294] [error] [t=27670] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    866866 type:Color
[2021-09-30 10:59:32.323] [error] [t=27670] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    900211 type:Color
[2021-09-30 10:59:32.347] [error] [t=27670] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    933544 type:Color
[2021-09-30 10:59:32.386] [error] [t=27670] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    966877 type:Color

[2021-09-30 10:59:32.422] [error] [t=27670] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1000222 type:Color
[2021-09-30 10:59:32.458] [error] [t=27670] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1033555 type:Color
[2021-09-30 10:59:32.490] [error] [t=27670] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1066866 type:Color
[2021-09-30 10:59:32.517] [error] [t=27670] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1100200 type:Color
[2021-09-30 10:59:32.556] [error] [t=27670] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1133533 type:Color
Segmentation fault (core dumped)
```

What is strange is that the following snippet works just fine:

```python
k4a = PyK4A(Config(color_resolution=pyk4a.ColorResolution.RES_720P, depth_mode=pyk4a.DepthMode.NFOV_UNBINNED,synchronized_images_only=True))
k4a.start()
capture = k4a.get_capture()
converted_image = capture.transformed_color
k4a.stop()
```

Firmware is version: 1.6.110079014 and the sdk version is 1.4.1.  OS is Ubuntu 18.04

Do you have any suggestions for how to handle this?  There is no onboard GPU, which may be the reason for the queue becoming full.

lpasselin, 2021-10-04:
Really weird! I'll have a look at this soon. Thanks for making the issue.

rajkundu, 2022-03-12:
Perhaps related to #159? I've had segmentation faults occur when the C SDK is passed the compressed/MJPEG color image rather than the decoded BGRA32 image it requires.

ekolve, 2022-04-19:
I wanted to update this issue in case anyone ever runs in to this.  We discovered that the USB-> barrel connector cable (power) to the Azure Kinect was plugged in to a 1A/5V USB port.  The docs state that the Azure Kinect can consume up to 5.9 watts.  When we plugged the cable into a AC USB adapter that can output 2.2 amps, we stopped getting segfaults.

closed 2022-04-19 by ekolve

mdabdk, 2022-07-14:
> I wanted to update this issue in case anyone ever runs in to this. We discovered that the USB-> barrel connector cable (power) to the Azure Kinect was plugged in to a 1A/5V USB port. The docs state that the Azure Kinect can consume up to 5.9 watts. When we plugged the cable into an AC USB adapter that can output 2.2 amps, we stopped getting segfaults.

In case anyone needs a reference to the "5.8 watts", [here](https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification#power-consumption) it is.