etiennedub/pyk4a #148: Time to wait for a frame

ShaoxiongYao, 2021-10-02:
Hi,

I am using Windows10 and SDK version 1.4.1. When I run the code,

`k4a = PyK4A(
        Config(
            color_resolution=pyk4a.ColorResolution.RES_720P,
            camera_fps=pyk4a.FPS.FPS_15,
            depth_mode=pyk4a.DepthMode.WFOV_2X2BINNED,
            synchronized_images_only=True,
        )
    )
    k4a.start()

    # getters and setters directly get and set on device
    k4a.whitebalance = 4500
    assert k4a.whitebalance == 4500
    capture_st_time = time.time()
        while True:
            capture = k4a.get_capture()
            if np.any(capture.depth) and np.any(capture.color):
                break
        print("capture_time:", time.time() - capture_st_time)`

It took about 0.8s to capture this frame, I wonder if this is normal?

shagren, 2021-10-04:
First frame usually have "big" delay.
Try to check delay between next frames.

closed 2021-10-04 by lpasselin

ShaoxiongYao, 2021-10-04:
Yes, the delay between next frames was fine, the delay can be as low as `0.01s` in `FPS_30` mode. Thanks!