etiennedub/pyk4a #151: kinect azure k4a.dll file is not found but I have installed the kinect SDK library in the C:\ folder

Qiu731, 2021-10-11:
I am sorry to bother you.I have checked all the answers about k4a.dll but it doesn't work.And I have installed the kinect SDK library in the C:\ folder.Are there any other possible reasons for the error？
[image]

chrismeunier, 2021-11-01:
> For some reason, I am unable to install version 1.2.4 (`pip install pyk4a==1.2.4`). It keeps saying "Failed building wheel for pyk4a". I guess I will have to wait for another way to fix.

Did you try to uninstall any current pyk4a installation and then use the _complete_ install command `pip install pyk4a==1.2.4 --no-use-pep517 --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"` ?

Even if the README now states that ""In most cases* `pip install pyk4a` *is enough*", I personally never succeeded installing with this standard install command.

sucharnt, 2021-11-02:
> Did you try to uninstall any current pyk4a installation and then use the _complete_ install command `pip install pyk4a==1.2.4 --no-use-pep517 --global-option=build_ext --global-option="-IC:\Program Files\Azure Kinect SDK v1.4.1\sdk\include" --global-option="-LC:\Program Files\Azure Kinect SDK v1.4.1\sdk\windows-desktop\amd64\release\lib"` ?
> 
> Even if the README now states that "_In most cases_ `pip install pyk4a` _is enough_", I personally never succeeded installing with this standard install command. Additionally this statement [has been added with v1.3.0](https://github.com/etiennedub/pyk4a/compare/1.2.4...1.3.0#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R35), so trying to install an earlier version of the package with this simpler command has no reason to work.

Work like a charm, thank you for the help!

closed 2021-11-11 by Qiu731