etiennedub/pyk4a #152: How to convert color map to depth map shape without attaching invalid depth point?

zhaocheng-hub, 2021-11-16:
Hello, how to convert color image to depth image size, but not append invalid depth image points to the image (those black points with depth value of 0)?

HannahHaensen, 2021-11-18:
Maybe like that: pyk4a.depth_image_to_color_camera(capture.depth, capture._calibration, capture.thread_safe)

zhaocheng-hub, 2021-11-19:
@HannahHaensen    Hello, the conversion diagram obtained through `pyk4a.color_image_to_depth_camera( ... )`  will still add depth failure points are black.   But I've got another way, thank you.

chrismeunier, 2021-12-13:
I would also suggest to use `pyk4a.color_image_to_depth_camera(color, depth, ...)`, the problem is that the depth image you give will always have missing points.
My workaround would be to create a fake depth image from raw IR sensor data. This works because the raw `capture.ir` arrays will almost never have blank pixels because of the sensor's noise.  

So here's my suggestion:

```python

import pyk4a
import cv2

k4a = pyk4a.PyK4A()
k4a.start()
capture = k4a.get_capture()
calib = k4a.calibration
ir_mask = capture.ir

for _ in range(3): # add some frames to be sure to have no blank pixels
    capture = k4a.get_capture()
    ir_mask += capture.ir
k4a.stop()

#Complete 'depth image' with no holes
depth_mask = np.ones(ir_mask.shape, dtype=np.uint16) * ir_mask.astype(bool) * np.iinfo(np.uint16).max
#Custom transformed image:
trans_clr_custom = pyk4a.color_image_to_depth_camera(capture.color, depth_mask, calib, True)
#Ref image:
trans_color = capture.transformed_color
#Result
cv2.imshow('color to depth', trans_color)
cv2.imshow('custom color to depth', trans_clr_custom)
cv2.waitKey(0)

```

Anyway @zhaocheng-hub  if you solved your problem, you could share your solution and you should close the issue.

zhaocheng-hub, 2021-12-14:
@chrismeunier  thank you

closed 2021-12-14 by lpasselin

chrismeunier, 2021-12-15:
Clearly using the mean depth to fill the blank points makes more sense!
For some added context: this script is based on something I made to get a mask of the blank corners of the depth sensor (that is `ir_mask.astype(bool)` but inverted). This adaptation gives a transformed color image without blanks, which answers the question, but it is far from ideal. Take ideas from this code, but do not reuse it blindly in some application.

> You could also fill in the blanks in `capture.depth` using a more complex method before performing the transformation.

☝️ this should be the clean way to go, filtering the depth image to fill the blanks with values that make sense would be a big improvement.