etiennedub/pyk4a #154: The meaning about the camera calibration got from the function "k4a.calibration_raw"

SeanHRI, 2021-12-12:
I connect the Azure Kinect Sensor by the fuction of "k4a.start()", I have not configure any setting such as the resolution, the function "k4a.calibration_raw" outputs the dict of many information.  The camera intrinsics matrix consists of 14 parameters but I think the number of parameters should be 9. Also, the resolution of image captured is 720 * 1280, but the SensorHeight and SensorWidth is either 1024 * 1024 or 4096 * 3072? May anyone else explain about these information? 
[image]

chrismeunier, 2021-12-13:
This is not an issue for this repository, you are asking a question while using the C++ wrapper of Microsoft on a python 3 wrapper project of the SDK...

I am not familiar with the C/C++ code but still here are some pointers:
- The raw calibration parameters you have in the second image are _raw_. These depend only on the camera and not on the configuration of the camera. They will _never_ change.
- The parameters you read from the opened Kinect and print in your console are linked to the current configuration of your Kinect. These will change for different configurations because while the sensors (color & IR) always have the same pixel size (3072x4096 & 1024x1024) the images' sizes and shapes depend on the resolution you set in your configuration.
- In more details: the parameters you obtained in your C++ code [as described here](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__calibration__intrinsic__parameters__t_1_1__param.html) depend on the current calibration. You can see you give the camera calibration, resolution and depth mode to `k4a_device_get_calibration()`. 
- Using `k4a.calibration_raw` with pyk4a gives you a result independent of the configuration of your device, it's just the _raw_ property, not the values corresponding to your current configuration.
- To get the processed values you should use `k4a.calibration.get_distortion_coefficients()` and `k4a.calibration.get_camera_matrix()` (see doc in [calibration.py](https://github.com/etiennedub/pyk4a/blob/905d77bfecc0ec425a6335dcddbd63cb09314334/pyk4a/calibration.py#L175)), these will give you the results you get with your C++ example.

closed 2021-12-13 by lpasselin

SeanHRI, 2021-12-14:
@chrismeunier I have tested and verified the results when using k4a.calibration.get_camera_matrix(), it is same to the results when using the k4a.dll in the visual studio 2019. Thank you for your detailed explanation.